mani-skill/ManiSkill #768: Questions about the opening cabinet task and joint physics

vlongle, 2024-12-24:
Hello,

Thank you for this great work! I'm looking at the [open_cabinet_drawer env](https://github.com/haosulab/ManiSkill/blob/main/mani_skill/envs/tasks/mobile_manipulation/open_cabinet_drawer.py). It seems that the `handle_link` is extracted by looking for `handle_` meshes within each link. 

We're interested in using our own automatically generated assets. The entire mesh of the object is generated by a diffusion model, which is then part-segmented. We would have a `drawer` link of the object but won't have fine-grained link like `handle`. Would ManiSkill/SAPIEN still work? Can the simulator rely on the fact that there is a protruding region of the drawer to simulate grasping motion without explicit `handle` annotation?

Secondly, the PartNet-Mobility assets do not have physics. Normally, in Mujoco, we'd have to do something like
```
<joint name="joint_2" axis="-1 0 0" type="slide" range="0 0.8" frictionloss="1" />
```
and randomize `frictionloss` for sim2real. But is physical properties handled in Maniskill? In your experience, do your default physic parameters transfer well to a real Franka arm?

vlongle, 2024-12-25:
Regarding the first question, I have made a version of the cabinet that doesn't rely on handle and tele-opted. Seems that the grasping physics still works as in the attached video.

https://github.com/user-attachments/assets/61b53837-e098-4848-b8e7-f1059a4c791c

StoneT2000, 2024-12-26:
For domain randomization see this: https://maniskill.readthedocs.io/en/latest/user_guide/tutorials/domain_randomization.html

Generally do all the domain randomizations during the load scene function.

I don't recall the partnet mobility dataset having material based labels. In MS1/MS2 we did domain randomization on joint properties like this by modifying friction and damping values

```
    def _set_joint_physical_parameters(self):
        for joint in self.cabinet.get_active_joints():
            joint.set_friction(...)
            joint.set_drive_property(
                stiffness=0, damping=...
            )
```
You can do something similar in MS3.