mani-skill/ManiSkill #769: [Question] How to Disable Collision Between the Gripper Fingers in ManiSkill?

yilin404, 2024-12-24:
I am a new user of ManiSkill, and I have been following the documentation to set up my robotic arm simulation. I used CoACD to perform convex decomposition on the original .stl files of the gripper model and converted it to the .glb format as instructed. However, I am encountering an issue: the gripper fingers are unable to open properly in the simulator, seemingly due to a collision between them.

In Sapien, I know that collisions between specific parts can be disabled by using the .srdf file. I was wondering if ManiSkill provides a similar method to disable the collision between the gripper fingers, or if there is an alternative way to address this issue (such as configuration settings or parameters that can be adjusted to prevent this collision detection).

To assist in diagnosing the issue, I have attached the mesh of the gripper as well as the corresponding URDF file for your reference.

Thank you very much for your time and support. I look forward to your guidance.
[Uploading a1arm_description_.zip…]()

StoneT2000, 2024-12-26:
Looks like you did it correctly (the python way) in your other issue #769! Closing for now

closed 2024-12-26 by StoneT2000

CreativeNick, 2024-12-27:
@yilin404 How did you end up using your .srdf file to disable specific collisions? I'm trying to do something very similar to fix a collision issue but my .srdf file doesn't seem to be disabling any parts.