mani-skill/ManiSkill #770: [Question] Unable to Pick Up Objects in the Simulator

yilin404, 2024-12-26:
I am facing an issue where the robotic arm in the simulator cannot pick up objects. The arm successfully moves to the target grasp position, closes the gripper, but the object remains stationary and is not lifted.
https://github.com/user-attachments/assets/159beff9-dd53-4e5b-ab5a-6d46b2e56d0e

The agent class is:
```
# ManiSkill
from mani_skill.agents.base_agent import BaseAgent, Keyframe
from mani_skill.agents.registration import register_agent
from mani_skill.agents.controllers import PDJointPosControllerConfig, PDJointPosMimicControllerConfig, \
                                          PDJointPosVelControllerConfig, \
                                          deepcopy_dict
from mani_skill.agents.utils import get_active_joint_indices
from mani_skill.utils import common, sapien_utils

import torch

# Typing
from mani_skill.utils.structs.actor import Actor

@register_agent()
class A1Arm(BaseAgent):
    uid = "a1arm"
    urdf_path = "./data/robot_description/a1arm_description/urdf/a1arm.urdf"
    urdf_config = dict(
        _materials=dict(
            gripper=dict(static_friction=2.0, dynamic_friction=2.0, restitution=0.1)
        ),
        link=dict(
            gripper1_axis=dict(
                material="gripper", patch_radius=0.1, min_patch_radius=0.1
            ),
            gripper2_axis=dict(
                material="gripper", patch_radius=0.1, min_patch_radius=0.1
            ),
        ),
    )

    fix_root_link = True
    load_multiple_collisions = True

    keyframes = dict()
    arm_joint_names = [
        "arm_joint1",
        "arm_joint2",
        "arm_joint3",
        "arm_joint4",
        "arm_joint5",
        "arm_joint6",
    ]
    gripper_joint_names = [
        "gripper1_axis", 
        "gripper2_axis"
    ]
    ee_link_name = "arm_seg6"

    arm_stiffness = 1e3
    arm_damping = 2e2
    arm_force_limit = 100.0

    gripper_stiffness = 1e3
    gripper_damping = 2e2
    gripper_force_limit = 100.0

    @property
    def _controller_configs(self):
        # -------------------------------------------------------------------------- #
        # Arm
        # -------------------------------------------------------------------------- #
        arm_pd_joint_pos = PDJointPosControllerConfig(
            self.arm_joint_names,
[... 233 lines cut ...]
                is_static = is_static,
                model_id = model_id,
                z_axis_protect=z_axis_protect,
                z_val_protect_value = z_axis_protect_value,
            )
        except:
            print(modelname, 'is not exsist model file!')
            return None, None

self.container, self.container_data = create_actor(
            self.scene,
            [0],
            container_pose,
            "002_container",
            model_id=np.random.choice(list(range(10))),
            z_axis_protect=True,
        )
        self.container.set_mass(0.01)
```

The object file is:
[7.zip](https://github.com/user-attachments/files/18249030/7.zip)

The object property is:
[image]

yilin404, 2024-12-26:
I sincerely apologize for the mistake in the code I previously uploaded. I've corrected the error and updated the repository with the modified code.
Unfortunately, it seems that the problem still persists even after the fix.

The agent class is:
```
# ManiSkill
from mani_skill.agents.base_agent import BaseAgent, Keyframe
from mani_skill.agents.registration import register_agent
from mani_skill.agents.controllers import PDJointPosControllerConfig, PDJointPosMimicControllerConfig, \
                                          PDJointPosVelControllerConfig, \
                                          deepcopy_dict
from mani_skill.agents.utils import get_active_joint_indices
from mani_skill.utils import common, sapien_utils

import torch

# Typing
from mani_skill.utils.structs.actor import Actor

@register_agent()
class A1Arm(BaseAgent):
    uid = "a1arm"
    urdf_path = "./data/robot_description/a1arm_description/urdf/a1arm.urdf"
    urdf_config = dict(
        _materials=dict(
            gripper=dict(static_friction=2.0, dynamic_friction=2.0, restitution=0.1)
        ),
        link=dict(
            gripper1=dict( # ----- I fix this error -----
                material="gripper", patch_radius=0.1, min_patch_radius=0.1
            ),
            gripper2=dict( # ----- I fix this error -----
                material="gripper", patch_radius=0.1, min_patch_radius=0.1
            ),
        ),
    )

    fix_root_link = True
    load_multiple_collisions = True

    keyframes = dict()
    arm_joint_names = [
        "arm_joint1",
        "arm_joint2",
        "arm_joint3",
        "arm_joint4",
        "arm_joint5",
        "arm_joint6",
    ]
    gripper_joint_names = [
        "gripper1_axis", 
        "gripper2_axis"
    ]
    ee_link_name = "arm_seg6"

    arm_stiffness = 1e3
    arm_damping = 2e2
    arm_force_limit = 100.0

    gripper_stiffness = 1e3
    gripper_damping = 2e2
    gripper_force_limit = 100.0

    @property
    def _controller_configs(self):
        # -------------------------------------------------------------------------- #
        # Arm
        # -------------------------------------------------------------------------- #
        arm_pd_joint_pos = PDJointPosControllerConfig(
            self.arm_joint_names,
[... 64 lines cut ...]
        l_contact_forces = self.scene.get_pairwise_contact_forces(
            self.gripper1_link, object
        )
        r_contact_forces = self.scene.get_pairwise_contact_forces(
            self.gripper2_link, object
        )
        lforce = torch.linalg.norm(l_contact_forces, axis=1)
        rforce = torch.linalg.norm(r_contact_forces, axis=1)

        # direction to open the gripper
        ldirection = self.gripper1_link.pose.to_transformation_matrix()[..., :3, 1]
        rdirection = -self.gripper2_link.pose.to_transformation_matrix()[..., :3, 1]
        langle = common.compute_angle_between(ldirection, l_contact_forces)
        rangle = common.compute_angle_between(rdirection, r_contact_forces)
        lflag = torch.logical_and(
            lforce >= min_force, torch.rad2deg(langle) <= max_angle
        )
        rflag = torch.logical_and(
            rforce >= min_force, torch.rad2deg(rangle) <= max_angle
        )
        return torch.logical_and(lflag, rflag)
```

The robot property is:
[image]

StoneT2000, 2024-12-26:
Can you share a video of what occurs?

And is the object you are picking up dynamic or static or kinematic?

yilin404, 2024-12-27:
The video is:
https://github.com/user-attachments/assets/59143fef-a155-49ad-b78d-321d9a9b849c

And the object is dynamic i think, becaure i use such code to create it `builder.set_physx_body_type("dynamic")`

StoneT2000, 2024-12-27:
I tried with a cube (density 1000, volume 0.04^3). 
[image]
Looks fine to me, lifted up correctly. It is possible the object you are trying to pick up is too heavy. If you really want to pick up something that is heavy you may need to modify controller.

Note is that if you are using position control, you should try to set the gripper position to a smaller value (e.g. -0.01 is okay as well, we do this for panda), it should try and exert more force to get a closed grip on the object.

Another note is that the gripper's collision mesh is overly complicated. Try to simplify it to a flat box shape. It may also improve the grasping, I didn't look too carefully but the gripper mesh could be somewhat wrong.

yilin404, 2024-12-30:
I’ve noticed another issue: in my code based on SAPIEN, I can easily grasp the object. Below are the parameter differences between the configurations in SAPIEN and ManiSkill:
The Gripper property in SAPIEN is:
[image]
The Gripper property in ManiSkill is:
[image]

The Object to pick property in SAPIEN is:
[image]
The Object to pick property in ManiSkill is:
[image]

yilin404, 2024-12-30:
I’ve noticed another issue: in my code based on SAPIEN, I can easily grasp the object. Below are the parameter differences between the configurations in SAPIEN and ManiSkill:
The Gripper property in SAPIEN is:
[image]
The Gripper property in ManiSkill is:
[image]

The Object to Pick property in SAPIEN is:
[image]
The Object to Pick property in ManiSkill is:
[image]

I would like to ask which of these parameters might significantly affect the grasping performance? Are there any specific ones I should prioritize when tuning?
Additionally, I plan to migrate the code to the ManiSkill framework for further experiments. Could you provide guidance on how to modify these parameters within ManiSkill?
Your insights would be highly valuable for narrowing down the potential cause. Thank you!

yilin404, 2024-12-31:
@StoneT2000

StoneT2000, 2025-02-03:
Can you try using the same contact_offset numbers?

StoneT2000, 2025-04-28:
Hi is there still an issue here?

closed 2025-04-28 by yilin404