mani-skill/ManiSkill #773: move slower with pd_ee_pose controller

Udntknwme, 2024-12-26:
Hi, I am using pd_ee_pose controller with panda franka to pick a cube. I move the tcp with env.step(target_action) by calculating target_action with the cube's pose in root frame. the target_action is fine, but the arm move too quick than expected.

is there any easy way to move slower, with pd_ee_pose controller. thx.

StoneT2000, 2024-12-26:
Try increasing the control frequency. The default sim / control frequency is 100:20, which means 5 physics steps (time worth 5/100 seconds) pass before you pass the next action (and render).

Alternative choice is to use substep rendering. If you use the record episode wrapper you can turn this on which then renders every subframe.

Udntknwme, 2024-12-27:
Thanks for your reply!

Sorry if I was not clear in my previous statement. The problem I'm currently running into is using the env.step(target_action) function to move tcp from pose1 to pose2. I'm calling the env.step(target_action)  function repeatedly with the same target_action in order to not block the movement of the simulation world. But tcp is moving too fast, for example the arm is moving too fast and throwing out the cube it is holding.

the code is like:
for _ in range(step):
    obs, reward, terminated, truncated, info = self.env.step(target_action)
    self.env.render()

and the behavior is like tcp reach target_action too fast and in much less than step, and wait for the step_num repeat finish.

I think what I need is maybe to set a velocity limit but I cannot find an appropiate function or attribute. 
Also I try to set env.agent.controller.configs['arm'].interpolate=True, and things seem to be better.

Is this setting right or is there any more suitable way to force the arm move slower? like move tcp from pose A to pose B in a set duration. Thx again!

StoneT2000, 2024-12-27:
It would be better for you to just interpolate between the current TCP pose and your target. Alternatively you can use a motion planning library to generate a sequence of targets.

The final option is to modify the agent controller. If you are using  PD EE delta pose, you can modify the max position and rotation changes allowed

closed 2025-01-28 by StoneT2000