mani-skill/ManiSkill #775: WARNING - No initial pose set for articulation builder

hesic73, 2024-12-27:
This warning occurs because `_load_agent` expects an `initial_agent_poses` argument, which defaults to None:

https://github.com/haosulab/ManiSkill/blob/42d4fcf84673a18c9446835eb55d888ade06bd1d/mani_skill/envs/sapien_env.py#L402

However, in `_reconfigure`, we cannot pass this parameter:

https://github.com/haosulab/ManiSkill/blob/42d4fcf84673a18c9446835eb55d888ade06bd1d/mani_skill/envs/sapien_env.py#L673

Is there a way to suppress this warning?

StoneT2000, 2024-12-27:
Every day I regret a little bit about making robots (which are articulations with controllers) managed by ManiSkill as opposed to just letting the users spawn one in.

The warning is just for you to help optimize the simulation. most times you can ignore it. To remove it add a def _load_agent(self, options) function like so: https://github.com/haosulab/ManiSkill/blob/main/mani_skill/envs/tasks/tabletop/pick_cube.py#L55-L56

Recommended to ensure the initial root pose is such that the robot doesn't collide with anything. Seems to be a limitation of physx at the moment to require initial poses as one needs to take one sim step to setup all the GPU buffers (and if that step causes collisions then there may be unnecessary overflows)

hesic73, 2024-12-27:
Thank you! Overloading `_load_agent` is both concise and elegant.

closed 2024-12-27 by hesic73