mani-skill/ManiSkill #778: [Question] Will there be more robot support?

KafuuChikai, 2024-12-28:
Hello,

I would like to know if there will be more robot support, such as **drones, autonomous vehicles**, etc. Additionally, will there be **more sensor support (e.g., LiDAR)**? Or are there any detailed tutorials available?

Thank you very much!

hesic73, 2024-12-28:
You can check out [OmniDrones](https://github.com/btx0424/OmniDrones) for drone support. It's based on Isaac Sim.

KafuuChikai, 2024-12-31:
Yes, I have used **OmniDrones** before, but **Isaac Sim** has many issues and is quite cumbersome to use in practice. That's why I'm here looking for alternative solutions. 😔

Thank you for your help!

StoneT2000, 2025-01-01:
@KafuuChikai I can chat with our team to see whether drone simulation will on our roadmap (currently it is not). Too many things to deal with in manipulation 😂

That being said if isaac isn't sufficient Genesis or Mujoco might be good. I think they both support Drone sim but I don't have domain knowledge about drone sim to comment on accuracy / speed.

KafuuChikai, 2025-01-02:
Thank you! I look forward to your future work! 😊

closed 2025-01-02 by KafuuChikai