mani-skill/ManiSkill #782: Background Rendering Behavior

hesic73, 2024-12-31:
I am working with an environment where I have a robot and an object, and there is nothing else in the scene. When I use the human-render mode, the background appears gray. However, when I use RGB rendering, the background appears dark. My question is: is there a way to set or configure the background color for RGB rendering? Thanks!

StoneT2000, 2025-01-01:
Change the sensors shader config from minimal to default:

See this: 
https://maniskill.readthedocs.io/en/latest/user_guide/concepts/sensors.html#shaders-and-textures

minimal is fast and low memory use, default is more flexible + background render. I haven't benchmarked the two shader configs before in depth but this is my experience with them.

Alternative you might be able to use an environment map which may still render in the minimal shader mode.

hesic73, 2025-01-01:
Thanks! This works for me.

closed 2025-01-01 by hesic73