mani-skill/ManiSkill #787: Randomness in GPU simulation

TritiumR, 2025-01-04:
Hi. I am wondering what could cause the sub-scene behave differently even if I use set_state() to set the environment to the same state and apply the same action in GPU simulation.

Is it due to the IK solver? Are there any way to set the seed or something to be the same to make all the sub-scene have the same results?

StoneT2000, 2025-01-05:
Generally setting seed should be nearly completely reproducible.

I am aware that for the renderer the same state can sometimes cause a extremely small difference in rendered images (eg 1-2 pixels out of 256x256)

Is there an example of this occurring, our tests usually check for this

TritiumR, 2025-01-06:
Thanks for the reply. I mean there is some divergence among the paralleled sub-scene as shown here. The states are set to the state of first sub-scene and the actions are same.

https://github.com/user-attachments/assets/c3aa08b6-ec80-45e6-ace6-6c54669267e6

StoneT2000, 2025-01-07:
Generally there is always some non-determinism due to likely floating point precisions causing small things to lead to larger divergences over longer periods of time. I think if you rerun the code it is reproducible within parallel env, but beyond not necessarily

Internally what really is happening is the e.g. object at position (x, y, z), is really in one big scene with position (x + sx, y + sy, z) where sx, sy is the origin of the subscene. There can be small minute differences here.

TritiumR, 2025-01-07:
Got it. Thanks a lot!

closed 2025-01-07 by TritiumR