mani-skill/ManiSkill #790: [Question] Motion Planning for more complex robots

CreativeNick, 2025-01-06:
I was looking at the documentation for [Motion Planning](https://maniskill.readthedocs.io/en/latest/user_guide/data_collection/motionplanning.html) and [mplib "Plan a Path" docs](https://motion-planning-lib.readthedocs.io/latest/tutorials/plan_a_path.html), and was wondering if motion planning would be recommended for a bimanual robot like the one below:

[image]

I'm writing a hand-off ppo where one hand picks up an randomly selected + positioned object and "hands it off" to the other hand. Given the many steps of this task, I was wondering if using motion planning would be an effective way of training the RL model to perform well. I'm concerned that the number of joints will and complexity of the task might render this method ineffective. What would be the best case for this task?

Thanks!

StoneT2000, 2025-01-07:
I recommend looking at the work from https://mihdalal.github.io/, particularly this work https://mihdalal.github.io/manipgen/

It's a neat way to combine motion planning with RL learned "local" control policies. In short it proposes using motion planning to get close to the object / goal pose (predicted via VLM or any other model), and then using a local control policy to finish the subtask (eg grasp ball, let go of ball). RL can be used to train that local control policy.

CreativeNick, 2025-01-07:
Thank you, I'll take a look!

closed 2025-01-09 by StoneT2000