mani-skill/ManiSkill #793: How to add a camera？

LittlePotatoChip, 2025-01-07:
I want to add a camera to hand_gripper,I tried to define a env "PickCubeWithCamera-v1" inherited from "PickCube-v1"

StoneT2000, 2025-01-07:
See https://maniskill.readthedocs.io/en/latest/user_guide/tutorials/custom_tasks/advanced.html#mounted-dynamically-moving-cameras for how to add mounted cameras in the environment class. You can mount to an actor or a link (eg a robot link).

If you have a custom robot you can take a look at the panda implementation we have which adds a wrist camera.

LittlePotatoChip, 2025-01-07:
> See https://maniskill.readthedocs.io/en/latest/user_guide/tutorials/custom_tasks/advanced.html#mounted-dynamically-moving-cameras for how to add mounted cameras in the environment class. You can mount to an actor or a link (eg a robot link).
> 
> The sensors / rendering system of ManiSkill is completely managed and we don't really recommend fetching it yourself with the low level APIs. Adding the camera sensors like above will have to returned in the observations dict.
> 
> If you have a custom robot you can take a look at the panda implementation we have which adds a wrist camera.

thanks for replying，I redefined an env "PickCubeCamera-v1" and an agent "panda_camera"：
```
import sapien
from mani_skill.utils import sapien_utils, common
from mani_skill.envs.sapien_env import BaseEnv
from mani_skill.envs.tasks.tabletop.pick_cube import PickCubeEnv
from mani_skill.utils.registration import register_env
from typing import Any, Dict, Union
from mani_skill.agents.robots import Fetch, Panda
from .panda_camera import PandaCamera

@register_env("PickCubeCamera-v1", max_episode_steps=50)
class PickCubeCameraEnv(PickCubeEnv):

    SUPPORTED_ROBOTS = ["panda", "fetch","panda_camera"]
    agent: Union[Panda, Fetch, PandaCamera]
    
    def __init__(self, *args, robot_uids="panda_camera", **kwargs):
        # robot_uids="fetch" is possible, or even multi-robot 
        # setups via robot_uids=("fetch", "panda")
        super().__init__(*args, robot_uids=robot_uids, **kwargs)

    def _load_agent(self, options: dict):
        super()._load_agent(options)

```

```
import numpy as np
from mani_skill.agents.base_agent import BaseAgent, Keyframe
from mani_skill.agents.robots.panda.panda import Panda
from mani_skill.agents.controllers import *
from mani_skill.agents.registration import register_agent
from mani_skill.sensors.camera import CameraConfig
from mani_skill import PACKAGE_ASSET_DIR

@register_agent()
class PandaCamera(Panda):
    uid = "panda_camera"

    urdf_path = f"{PACKAGE_ASSET_DIR}/robots/panda/panda_v3.urdf"

    @property
    def _sensor_configs(self):
        return [
            CameraConfig(
                uid="panda_hand_camera",
                pose=sapien.Pose(p=[0, 0, 0], q=[1, 0, 0, 0]),
                width=128,
                height=128,
                fov=np.pi / 2,
                near=0.01,
                far=100,
                mount=self.robot.links_map["camera_link"],
            )
        ]

```
and make a env by:

```
env = gym.make(
    "PickCubeCamera-v1", # there are more tasks e.g. "PushCube-v1", "PegInsertionSide-v1", ...
    num_envs=1,
    obs_mode="rgb+depth", # there is also "state_dict", "rgbd", ...
    control_mode="pd_ee_delta_pose", # there is also "pd_joint_delta_pos", ...
    render_mode="human",
    robot_uids = "panda_camera"
)
```

camera can be rendered and the rbg+depth data can be captured from obs_dict.it works!!

But I found that the initial joint variables of Panda have changed.
that's pickcube-v1:
[image]
and that's my env:
[image]

I think I did't change too many things, but what cause it?can I change this?

StoneT2000, 2025-01-08:
This is because you define a new robot called panda_camera. The original pick cube environment uses the table scene builder which sets the initial pos of the robot based on the given robot id. I recommend in the `_initialize_episode` function to set the panda initial qpos yourself to whatever you like.

LittlePotatoChip, 2025-01-08:
> panda initial qpos yourself to whatever you like.

thanks,I'll read the doc again.

closed 2025-01-08 by LittlePotatoChip