mani-skill/ManiSkill #799: 3D Diffusion Policy Implementation

ruoxianglee, 2025-01-14:
Hi, I am trying to use 3d diffusion policy (DP3) code in ManiSkill, but encountering some issue. I would appreciate your help with this matter.

The DP3 code has been added into baselines [diffusion_policy_3d](https://github.com/ruoxianglee/ManiSkill/tree/main/examples/baselines/diffusion_policy_3d). I trained DP3 for PickCube task in ManiSkill environment. Druing training, it was observed that the loss can converge very fast to a small value. Unfortunately, The success rate during evaluation is very low. The highest success rate I got is 10%. It is close to 0 for most cases. I tried to find the potential reason but failed.

To implement DP3 into ManiSkill environment, I did the following modifications:
- Convert the ManiSkill demonstrations from hdf5 format into zarr format ([see code here](https://github.com/ruoxianglee/ManiSkill/tree/main/examples/baselines/diffusion_policy_3d/dataset/convert_hdf5_to_zarr.py)), so the demonstrations can be directly used for training DP3 policy.
- ManiSkill dataset for pytorch datasetloader ([see code here](https://github.com/ruoxianglee/ManiSkill/tree/main/examples/baselines/diffusion_policy_3d/diffusion_policy_3d/dataset/maniskill_dataset.py))
- ManiSkill environment [wrapper](https://github.com/ruoxianglee/ManiSkill/tree/main/examples/baselines/diffusion_policy_3d/diffusion_policy_3d/env/maniskill_wrapper.py) and [runner](https://github.com/ruoxianglee/ManiSkill/tree/main/examples/baselines/diffusion_policy_3d/diffusion_policy_3d/env_runner/maniskill_runner.py)

The followings are the key settings:
- TaskID: PickCube-v1
- Observation mode: pointcloud (pointcloud cropping and downsampling are used)
- Controller: arm_pd_ee_delta_pose
- Simulation backend: cpu
- Number of demos: 10
- Observation
  - State: 29-D (including joint positions [7+2], velocities [7+2], goal position [3], end-effector pose [7], is_grasped [1])
  - Pointcloud: 6-D (xyz [3], rgb [3])
- Action: 7-D (translation [3], rotation [3], gripper open/close [1])

StoneT2000, 2025-01-28:
Thanks for contributing the DP3 baseline! Are there issues still for this? I have a bit more time now to help take a look. I always recommend to first test on PushCube before anything else. PickCube sounds easy but its actually a harder task than tasks in a lot of other benchmarks (e.g. D4RL or metaworld) due to the amount of randomization

ruoxianglee, 2025-02-04:
Hi @StoneT2000 , thanks for your suggestion. I tested PushCube task with 100 demos. The success rate is 100. Based on this, I think your insight is totally correct and the code works well. I will try to test other tasks, e.g., PickCube, and will post the result here. Thanks.

linden713, 2025-02-23:
Hi @ruoxianglee . Thank you for your great work! I think ManiSkill is a great platform, and I also want to run DP3 on it. Your deployment helped me a lot. Have you tried other tasks? How did they go?

mkygogo, 2025-05-11:
Hi@ruoxianglee , I tried your DP3 , I got error like this :
$ python dataset/convert_hdf5_to_zarr.py --num_demos 10 --zarr_dir ./data/ --hdf5_path ~/maniskill/ManiSkill/demos/PickCube-v1/motionplanning/trajectory_cpu.pointcloud.pd_ee_delta_pose.physx_cpu.h5 --env_name PickCube-v1
Number of demos/episodes : 10
Loading HDF5 file /home/jr/maniskill/ManiSkill/demos/PickCube-v1/motionplanning/trajectory_cpu.pointcloud.pd_ee_delta_pose.physx_cpu.h5
Loaded
Load observations 10 <class 'dict'>
Load actions 10 <class 'numpy.ndarray'>
Traceback (most recent call last):
  File "/home/jr/maniskill/ManiSkill/examples/baselines/diffusion_policy_3d/dataset/convert_hdf5_to_zarr.py", line 198, in <module>
    main(args)
  File "/home/jr/maniskill/ManiSkill/examples/baselines/diffusion_policy_3d/dataset/convert_hdf5_to_zarr.py", line 124, in main
    obs_robot_state = common.flatten_state_dict(
  File "/home/jr/anaconda3/envs/maniskill-dp3/lib/python3.9/site-packages/mani_skill/utils/common.py", line 219, in flatten_state_dict
    if state.nelement() == 0:
AttributeError: 'numpy.ndarray' object has no attribute 'nelement'

How can I fix it ? thanks

yqi19, 2025-07-03:
Hi, @ruoxianglee @mkygogo May I ask does the code of dp3 released? I can not see it yet.

linden713, 2025-07-03:
Hi @yqi19,  you can try the code [here ](https://github.com/linden713/ManiSkill/tree/main/examples/baselines/diffusion_policy_3d)and most modifications are made by @ruoxianglee.
I made some changes based on my task, so you may need to git revert what I committed

yqi19, 2025-07-03:
Thank you very much !! 🧡