mani-skill/ManiSkill #800: Problem with Setting Friction Parameters for an Actor

SumeetBatra, 2025-01-14:
Hi all. Following the tutorial you all provided, I'm trying to do some simple physics domain randomizations of the cube in the PickCube task like so:

            for obj in self.cube._objs:
                # modify cube friction
                dyn_friction_range = self.domain_rand_cfg['cube']['dyn_friction_range']
                static_friction_range = self.domain_rand_cfg['cube']['static_friction_range']
                for shape in obj.components[1].collision_shapes:
                    shape.physical_material.dynamic_friction += np.random.uniform(
                    dyn_friction_range[0], dyn_friction_range[1])
                    shape.physical_material.static_friction += np.random.uniform(
                    static_friction_range[0], static_friction_range[1])

However when I do this, it seems that setting the friction values for one object in self.cube._objs sets the same values for all the actor objects. Is this a bug or am I doing something wrong? Thanks!

StoneT2000, 2025-01-21:
This sounds like a bug. Could you help print the object id 

```
# something like this
print(id(shape), id(shape.physical_material))
```

SumeetBatra, 2025-02-10:
This is what gets printed:
```

id(shape)=130444034670448, id(shape.physical_material)=130444034737904
id(shape)=130444034736944, id(shape.physical_material)=130444034737904
id(shape)=130444034670448, id(shape.physical_material)=130444034737904
id(shape)=130444034736944, id(shape.physical_material)=130444034737904
id(shape)=130444034670448, id(shape.physical_material)=130444034737904
id(shape)=130444034736944, id(shape.physical_material)=130444034737904
id(shape)=130444034670448, id(shape.physical_material)=130444034737904
id(shape)=130444034736944, id(shape.physical_material)=130444034737904
```

StoneT2000, 2025-02-11:
Ah okay. I think the solution here might just be that you need to create each cube with a separate sapien.physx.PhysicalMaterial object attached. Let me get back to you on what's the easiest way to do that. Documentation needs to be updated on this as well, seems some properties cannot be easily changed like that.

StoneT2000, 2025-03-07:
https://maniskill.readthedocs.io/en/latest/user_guide/tutorials/domain_randomization.html#actor-link-physical-and-visual-randomizations has new updated guide on how to do randomizations in maniskill. Key part is you need to basically build each cube separately and then merge them together into one view/object. Building separately enables the creation of separate materials (which normally is shared for memory optimization purposes and to stay under certain physx limits like max physx materials)