mani-skill/ManiSkill #804: Question about GAE normalization strategy in PPO implementation

songyuc, 2025-01-24:
Hello,

I am a student currently learning about reinforcement learning and I came across your PPO implementation in the ManiSkill repository. I noticed a particular comment in the code that mentions normalizing the GAE by the sum of `lambda^i` instead of the standard 1-lambda method.

Here is the relevant [code-snippet](https://github.com/haosulab/ManiSkill/blob/main/examples/baselines/ppo/ppo.py#L357-L371).
I am curious about the reasoning behind this approach. Is there a specific advantage or scenario in which this method is preferred? I would greatly appreciate any insights or references to related literature.

Thank you for your time and for providing such a comprehensive resource!

Best regards,
Yucheng Song

StoneT2000, 2025-02-13:
This is just for the finite horizon GAE implementation originally by @tongzhoumu. We used to use it for our CPU sim environments and found it to work relatively well. For GPU sim it doesn't work as well (since we typically take like a few steps per environment per rollout and it is better to model the original implementation)

tongzhoumu, 2025-02-13:
Hi @songyuc 

The original GAE paper assumes an infinite horizon but this does not apply to most episodic tasks we use nowadays. Most implementations just truncated the formula but it does not make too much sense mathematically. Our finite horizon GAE implementation probably aligns better with the original infinite horizon GAE from a math perspective, but it works similarly to the naive truncation version empirically.

closed 2025-02-18 by StoneT2000