mani-skill/ManiSkill #805: ProcTHOR static object poses are all zero

BlGene, 2025-01-24:
I've found a workaround for this issue, so it's not urgent.

Hi @StoneT2000 

I was trying to get the poses of items in ProcTHOR, however these were returning only zeros.
(via `procthor_scene.scene_objects[object_name].pose.get_p() `)

I tried to fix it by setting a value for `builder.initial_pose = sapien.Pose()` however, this messed the scene up. 

It looks like the debug viewer is also giving zero values for the poses. (see below)

[image]

The workaround for me was to add a `self._default_scene_objects_poses.append((actor, pose))` to the scene builder. Could provide a PR if it's helpful.

StoneT2000, 2025-01-28:
Is this on the GPU sim or CPU sim?

BlGene, 2025-01-31:
Hi @StoneT2000, 

Yes sure! The output of ` env.print_sim_details()` is :
```
# -------------------------------------------------------------------------- #
Task ID: ClevrMove-v1, 1 parallel environments, sim_backend=cpu
obs_mode=rgb+depth+segmentation, control_mode=pd_joint_pos
render_mode=rgb_array, sensor_details=RGBD(448x448), RGBD(128x128), RGBD(128x128)
sim_freq=100, control_freq=20
observation space: Dict('agent': Dict('qpos': Box(-inf, inf, (1, 15), float32), 'qvel': Box(-inf, inf, (1, 15), float32)), 'extra': Dict('tcp_pose': Box(-inf, inf, (1, 7), float32)), 'sensor_param': Dict('render_camera': Dict('extrinsic_cv': Box(-inf, inf, (1, 3, 4), float32), 'cam2world_gl': Box(-inf, inf, (1, 4, 4), float32), 'intrinsic_cv': Box(-inf, inf, (1, 3, 3), float32)), 'fetch_head': Dict('extrinsic_cv': Box(-inf, inf, (1, 3, 4), float32), 'cam2world_gl': Box(-inf, inf, (1, 4, 4), float32), 'intrinsic_cv': Box(-inf, inf, (1, 3, 3), float32)), 'fetch_hand': Dict('extrinsic_cv': Box(-inf, inf, (1, 3, 4), float32), 'cam2world_gl': Box(-inf, inf, (1, 4, 4), float32), 'intrinsic_cv': Box(-inf, inf, (1, 3, 3), float32))), 'sensor_data': Dict('render_camera': Dict('depth': Box(-32768, 32767, (1, 448, 448, 1), int16), 'rgb': Box(0, 255, (1, 448, 448, 3), uint8), 'segmentation': Box(-9223372036854775808, 9223372036854775807, (1, 448, 448, 1), int64)), 'fetch_head': Dict('depth': Box(-32768, 32767, (1, 128, 128, 1), int16), 'rgb': Box(0, 255, (1, 128, 128, 3), uint8), 'segmentation': Box(-9223372036854775808, 9223372036854775807, (1, 128, 128, 1), int64)), 'fetch_hand': Dict('depth': Box(-32768, 32767, (1, 128, 128, 1), int16), 'rgb': Box(0, 255, (1, 128, 128, 3), uint8), 'segmentation': Box(-9223372036854775808, 9223372036854775807, (1, 128, 128, 1), int64))))
(single) action space: Box([-1.606 -1.221   -inf -2.251   -inf -2.16    -inf -1.    -1.    -1.
 -1.    -1.    -1.   ], [1.606 1.518   inf 2.251   inf 2.16    inf 1.    1.    1.    1.    1.
 1.   ], (13,), float32)
# -------------------------------------------------------------------------- #
```

StoneT2000, 2025-02-06:
Ok I think this is because the way the env is designed is a bit off. The relative pose of the visual meshes are set to the object pose instead of the object origin being set the object pose. I can fix this now

Also what is `_default_scene_objects_poses`, i don't see this in our code?

StoneT2000, 2025-02-06:
Actually is it just the static objects that have pose 0? I think that is what I get on my end. It seems the code is not setting the static object initial poses

StoneT2000, 2025-02-06:
Ok should be fixed soon, making a PR now.

StoneT2000, 2025-02-06:
Note that your workaround is not quite correct, it might cause the objects to look like in the wrong place despite the object.pose saying its in the right place (and it will fail in GPU sim). Please use the latest commit on the main branch which fixes this (or just copy the fix from #836 )

PR #836 (2025-02-06): [BugFix] Fix initial poses of static objects in AI2THOR scenes to be not zero in CPU and GPU sim.
Previously the code initialized the meshes at the pose of the object instead of the whole object at that pose. Closes #805
closed 2025-02-06 by StoneT2000