mani-skill/ManiSkill #807: Where does the env registeration happens for ai2thor and robocasa envs

Felix-Zhenghao, 2025-01-27:
I can only find the scene_builder code. But where is the environment code? You call gym.make so the environments are registered, but I can't find where they are registered.

Felix-Zhenghao, 2025-01-27:
Also, if I set breakpoint at the `build()` of ai2thor's scene_builder, it just does not break anything. But when I check the env type that `gym.make` returns, it is heavily wrapped but there is actually an ai2thor scene_builder inside `env.env.env`. So I 'm wondering what's happening under the hood, but I can't find the code.

StoneT2000, 2025-01-28:
The the scene builder is primarily meant for users to build on top of / use directly in their own custom environments.

If you want to look at the specific code using it there is a generic scene exploration environment https://github.com/haosulab/ManiSkill/blob/main/mani_skill/envs/scenes/base_env.py

closed 2025-01-28 by Felix-Zhenghao