mani-skill/ManiSkill #843: [Question] Export Visual Mesh with Texture/Color

hesic73, 2025-02-06:
I want to export the visual mesh of an actor from ManiSkill3 (Sapien). I found this function:

https://github.com/haosulab/ManiSkill/blob/5887500f11e5dd405626f4a8ec659d47139082d3/mani_skill/utils/geometry/trimesh_utils.py#L100-L104

However, it appears that the underlying implementation only returns vertices and faces without carrying any texture or color information:

https://github.com/haosulab/ManiSkill/blob/5887500f11e5dd405626f4a8ec659d47139082d3/mani_skill/utils/geometry/trimesh_utils.py#L55-L65

Looking into Sapien’s `RenderShapeTriangleMeshPart`, it seems that it does not provide the necessary information either:

```python
class RenderShapeTriangleMeshPart:
    __hash__: typing.ClassVar[None] = None
    def __eq__(self, arg0: RenderShapeTriangleMeshPart) -> bool:
        ...
    def get_cuda_triangles(self) -> sapien.pysapien.CudaArray:
        ...
    def get_cuda_vertices(self) -> sapien.pysapien.CudaArray:
        ...
    def get_material(self) -> RenderMaterial:
        ...
    def get_triangles(self) -> numpy.ndarray[numpy.uint32[M, 3]]:
        ...
    def get_vertex_normal(self) -> numpy.ndarray[tuple[M, typing.Literal[3]], numpy.dtype[numpy.float32]]:
        ...
    def get_vertex_uv(self) -> numpy.ndarray[tuple[M, typing.Literal[2]], numpy.dtype[numpy.float32]]:
        ...
    def get_vertices(self) -> numpy.ndarray[tuple[M, typing.Literal[3]], numpy.dtype[numpy.float32]]:
        ...
    def set_vertex_normal(self, normal: numpy.ndarray[tuple[M, typing.Literal[3]], numpy.dtype[numpy.float32]] | list | tuple) -> None:
        ...
    def set_vertex_uv(self, uv: numpy.ndarray[tuple[M, typing.Literal[2]], numpy.dtype[numpy.float32]] | list | tuple) -> None:
        ...
    @property
    def cuda_triangles(self) -> sapien.pysapien.CudaArray:
        ...
    @property
    def cuda_vertices(self) -> sapien.pysapien.CudaArray:
        ...
    @property
    def material(self) -> RenderMaterial:
        ...
    @property
    def triangles(self) -> numpy.ndarray[numpy.uint32[M, 3]]:
        ...
    @property
    def vertices(self) -> numpy.ndarray[tuple[M, typing.Literal[3]], numpy.dtype[numpy.float32]]:
        ...
```

Given this, is there a way to export a mesh from ManiSkill3/Sapien with texture/color information included? Thanks!

StoneT2000, 2025-02-12:
@fbxiang any idea for how to do this with sapien

pvskand, 2025-02-13:
@fbxiang has written code to sample colored points from sapien meshes [here](https://github.com/fbxiang/sapien-sample-points/tree/main). You will have to use the `sample_sapien_mesh` [function](https://github.com/fbxiang/sapien-sample-points/blob/main/sample_mesh.py#L47C5-L47C23).

In the arguments it's written that the `sample_sapien_mesh` takes in `sapien.render.RenderShapeTriangleMesh` object but I found that it actually returns an error saying `*** AttributeError: 'sapien.pysapien.render.RenderShapeTriangleMeshPart' object has no attribute 'parts'` so you will have to send in one of `{RenderShapeBox, RenderShapeCylinder, RenderShapeSphere}` objects.

Specifically, you can call the `sample_sapien_mesh(render_shape, N=1000)` [right before L51](https://github.com/haosulab/ManiSkill/blob/e2954732ba37c7f38816b3d038a0aa41da4275b3/mani_skill/utils/geometry/trimesh_utils.py#L51) of the `get_render_body_meshes()` function. 

This will [return you a dictionary](https://github.com/fbxiang/sapien-sample-points/blob/main/sample_mesh.py#L146-L151) of sampled point cloud along with its normals, color and material properties. However, if you'd like to get the mesh color, I believe as long as the mesh has a uniform color -- the following re-write of the `get_render_shape_meshes()` [function](https://github.com/haosulab/ManiSkill/blob/e2954732ba37c7f38816b3d038a0aa41da4275b3/mani_skill/utils/geometry/trimesh_utils.py#L55-L65) should work:

```
for part in rendered_shape.parts:
    vertices = part.vertices * render_shape.scale  # [n, 3]
    faces = part.triangles
    color = part.material.base_color # should return the base color
    mesh = trimesh.Trimesh(vertices=vertices, faces=faces) 
    mesh.visual = trimesh.visual.color.ColorVisuals(mesh, [1, 1, 0]) # set the colors
    ...
```

I haven't tested this in detail but from my preliminary testing it turns out fine. Please post here if you run into issues.

hesic73, 2025-02-15:
Thank you for your detailed response! Actually, I had a sudden thought—would it be possible to directly export YCB models from ManiSkill via an API? I can see that they are stored in `~/.maniskill`, and YCB also provides a download link for the 3D models: [YCB Models](http://ycb-benchmarks.s3-website-us-east-1.amazonaws.com/). I understand now.

closed 2025-02-15 by hesic73

StoneT2000, 2025-02-15:
The ycb models used are all in the data folder, you can write a script to iterate over each .obj file in there