mani-skill/ManiSkill #849: The problem of acquiring contact points between a robot and an object.

YZY14606, 2025-02-15:
During the simulation, we used get_contacts() and get_pairwise_contacts() to acquire the contact points.However, when we performed visualization to verify the contact points, we found that the acquired contact points were not actual contact points. The points we retrieved were only on the object, not on the robot.
We want to obtain the contact point information between the panda robot's fingers and the object.
The point cloud visualization image is as follows (The red points are contact points acquired. The green points are scene points.):

[image]

The RGB visualization image is as follows:

[image]

Has anyone encountered this issue before? Could it be related to the simulator? Or how do others typically obtain contact point information?

StoneT2000, 2025-02-15:
CPU or GPU simulation?

And what module is the function from?

YZY14606, 2025-02-15:
My num_envs=1, and it should be CPU simulation?

The function `get_contacts()` comes form the `mani_skill.envs.scene` module, and the code in my task are as follows:
`all_contacts = env.unwrapped.scene.get_contacts()`.

The function `get_pairwise_contacts()` comes from `mani_skill.utils.sapien_utils module` and the code in my task are as follows:
`bowl_effector_contact = get_pairwise_contacts(all_contacts, env.unwrapped.obj._bodies[0].entity, env.unwrapped.agent.finger2_link._bodies[0].entity)`.

YZY14606, 2025-02-15:
@StoneT2000 
There may be an issue with incorrect contact detection.I reproduced the scene using consecutive frames of RGB images and point cloud data.
The RGB images are as follows:
[image]

[image]

[image]
The visualized point cloud is as follows:

[image]

The visualized point cloud image corresponds to the third frame of the RGB sequence. In this image, the red point is identified as contact point. However, based on the RGB images from the consecutive frames, it appears that the robot's fingers are not actually in contact with the object.

StoneT2000, 2025-02-15:
Ah so that function get_contacts if i remember correctly returns all potential contacts where objects are within contact_offset distance of each other.

scene.get_pairwise_contact_forces is the function you want to use, pass the gripper link and actor and if the force is non zero then there is contact

YZY14606, 2025-02-16:
@StoneT2000 
Thanks very much for your reply! 
But I still have a question:
If the `get_contacts` function has been deprecated, which function can we use to get the contact point position?

YZY14606, 2025-02-16:
@StoneT2000 
And there still is a problem. Although we have excluded the moments of false contacts through the `scene.get_pairwise_contact_forces` function, due to the `contact_offset distance`, we will still get some points incorrectly identified as contacts in the real collision. Can we adjust the value of `contact_offset distance` to achieve higher contact point accuracy?

StoneT2000, 2025-02-19:
Are you saying scene.get_pairwise_contact_forces is returning forces when there are none?

And for get_contacts, that is `physx_cpu` sim_backend only feature at the moment. It returns all potential contacts and you have to filter out by which have nonzero impulses.

On `physx_cuda` sim_backend (the GPU sim) we currently do not support returning the exact contact points.

YZY14606, 2025-02-20:
Okay, I got it. Thanks very much for your reply again!

closed 2025-02-21 by YZY14606