mani-skill/ManiSkill #850: Error: Replay with different control modes are not supported when replaying on GPU parallelized environments

yijie21, 2025-02-15:
As I use GPU to replay the trajectories with the following command provided in the README.md

`python -m mani_skill.trajectory.replay_trajectory \
  --traj-path ~/.maniskill/demos/PickCube-v1/motionplanning/trajectory.h5 \
  --use-first-env-state -c pd_ee_delta_pos -o state \
  --save-traj --num-procs 1 -b gpu --count 100 # process only 100 trajectories`

It reports the following error:

`NotImplementedError: Replay with different control modes are not supported when replaying on GPU parallelized environments`

How to solve this problem?

StoneT2000, 2025-02-18:
Documentation is outdated, we recently revamped the trajectory replay tool to replay gpu and cpu trajectories more flexibly 

We do not support at the moment action space / controller conversion on the GPU as the batched nature makes some conversions a bit complicated.

What you can do is replay with cpu backend first converting the action space, then if you want to check if the trajectory works on the physx_cuda backend (what backend gpu refers to now), then run the script you shared. The use case for the script you use is primarily to check if the physx_cpu trajectory actions replay fine on the other backend (which doesn't work for highly precise or dynamic tasks due to floating precision problems)

yijie21, 2025-02-20:
Thanks for your reply. Actually I was testing the examples in this repo such as `act`, however, I can replay the trajectory with your provided command:

`python -m mani_skill.trajectory.replay_trajectory \
  --traj-path ~/.maniskill/demos/PickCube-v1/motionplanning/trajectory.h5 \
  --use-first-env-state -c pd_ee_delta_pos -o state \
  --save-traj --num-procs 10`
(where I need to remove `--num-procs 10` to make this command work)

But as I execute the following command:

`python train.py --env-id PickCube-v1 \
  --demo-path ~/.maniskill/demos/PickCube-v1/motionplanning/trajectory.state.pd_ee_delta_pos.cuda.h5 \
  --control-mode "pd_ee_delta_pos" --sim-backend "cpu" --num-demos 100 --max_episode_steps 100 \
  --total_iters 30000 `

It reports that I there is no `trajectory.state.pd_ee_delta_pos.cuda.h5`, then I check the generated trajectory file, it is named `trajectory.state.pd_ee_delta_pos.physx_cpu.h5`. Therefore, I change the name to `trajectory.state.pd_ee_delta_pos.physx_cpu.h5`, but it reports:

`RuntimeError: Cannot set the sim backend to 'cpu' and have multiple environments.
            If you want to do CPU sim backends and have environment vectorization you must use multi-processing across CPUs.
            This can be done via the gymnasium's AsyncVectorEnv API`

So I wonder how to solve this? And how to run the examples such as `act` and `diffusion_policy` without reporting errors?

StoneT2000, 2025-02-20:
I think you might be using some outdated scripts. I recommend following https://maniskill.readthedocs.io/en/latest/user_guide/learning_from_demos/setup.html to get an understanding of how to setup imitation learning workflows.

The official trajectory replay scripts for imitation learning benchmarking is here now: https://github.com/haosulab/ManiSkill/blob/main/scripts/data_generation/replay_for_il_baselines.sh

The latest baselines.sh files in `act` and `diffusion_policy` use data replayed by that script.

I will need to update the README's for act and diffusion policy, we used to write each time for each baseline algorithm how to replay data but we have moved that information away to the setup.html page i linked earlier, and have consolidated all the replay scripts to be the same so that everyone uses the same datasets, same action space etc.

yijie21, 2025-02-22:
Thank you very much for this quick reply!!!! You really saved me from being confused.

And looking forward to the official release of ManiSkill3.

StoneT2000, 2025-02-23:
Yeah "official" release out of beta currently is just pending our reproducible sim2real manipulation project and VR teleoperation code being merged in. After that updates will be more stable

closed 2025-02-23 by StoneT2000