mani-skill/ManiSkill #851: [Bug] Simply creating a jax array breaks rendering

nv-quan, 2025-02-17:
I'm trying to integrate a jax project with maniskill. However I found that simply creating a jax array would break maniskill's rendering. Here is a small example:
```python
import gymnasium as gym
import mani_skill.envs
import jax
import jax.numpy as jnp

num_envs = 2

env = gym.make(
    "PushCube-v1",
    obs_mode="state",
    control_mode="pd_joint_delta_pos",
    num_envs=num_envs,
    render_mode="human",
    parallel_in_single_scene=True,
)

obs, _ = env.reset()
test = jnp.zeros(2) #Everything works fine after this line is deleted
action = env.action_space.sample()
obs, reward, terminated, truncated, info = env.step(action)
env.render()
```
Then do installation:
```bash
pip install --upgrade mani_skill
pip install torch
pip install -U "jax[cuda12]"
```
Here is the error I got:

```bash
Traceback (most recent call last):
  File "test_env.py", line 21, in <module>
    env.render()
  File "/home//.pyenv/versions/maniskill_bug/lib/python3.11/site-packages/gymnasium/core.py", line 471, in render
    return self.env.render()
           ^^^^^^^^^^^^^^^^^
  File "/home//.pyenv/versions/maniskill_bug/lib/python3.11/site-packages/gymnasium/wrappers/order_enforcing.py", line 70, in render
    return self.env.render(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home//.pyenv/versions/maniskill_bug/lib/python3.11/site-packages/mani_skill/envs/sapien_env.py", line 1285, in render
    return self.render_human()
           ^^^^^^^^^^^^^^^^^^^
  File "/home//.pyenv/versions/maniskill_bug/lib/python3.11/site-packages/mani_skill/envs/sapien_env.py", line 1220, in render_human
    self._viewer.render()
  File "/home//.pyenv/versions/maniskill_bug/lib/python3.11/site-packages/sapien/utils/viewer/viewer.py", line 225, in render
    self.window.render(self.render_target, ui_windows)
RuntimeError: cannot create buffer
```
If I delete the line `test = jnp.zeros(2) ` everything works again.

Additional info:
```
Python version: 3.11.10
GPU: NVIDIA GeForce RTX 4060
Ubuntu 22.04
```
`pip freeze`
```bash
absl-py==2.1.0
arm_pytorch_utilities==0.4.3
asttokens==3.0.0
certifi==2025.1.31
charset-normalizer==3.4.1
cloudpickle==3.1.1
contourpy==1.3.1
cycler==0.12.1
dacite==1.9.2
decorator==5.1.1
docstring_parser==0.16
executing==2.2.0
Farama-Notifications==0.0.4
fast_kinematics==0.2.2
filelock==3.17.0
fonttools==4.56.0
fsspec==2025.2.0
gitdb==4.0.12
GitPython==3.1.44
gymnasium==0.29.1
h5py==3.12.1
huggingface-hub==0.28.1
idna==3.10
imageio==2.37.0
imageio-ffmpeg==0.6.0
ipython==8.32.0
jax==0.5.0
jax-cuda12-pjrt==0.5.0
jax-cuda12-plugin==0.5.0
jaxlib==0.5.0
jedi==0.19.2
Jinja2==3.1.5
kiwisolver==1.4.8
lxml==5.3.1
mani_skill==3.0.0b18
markdown-it-py==3.0.0
MarkupSafe==3.0.2
matplotlib==3.10.0
matplotlib-inline==0.1.7
mdurl==0.1.2
ml_dtypes==0.5.1
mplib==0.1.1
mpmath==1.3.0
networkx==3.4.2
numpy==1.26.4
nvidia-cublas-cu12==12.4.5.8
nvidia-cuda-cupti-cu12==12.4.127
nvidia-cuda-nvcc-cu12==12.8.61
nvidia-cuda-nvrtc-cu12==12.4.127
nvidia-cuda-runtime-cu12==12.4.127
nvidia-cudnn-cu12==9.1.0.70
nvidia-cufft-cu12==11.2.1.3
nvidia-curand-cu12==10.3.5.147
nvidia-cusolver-cu12==11.6.1.9
nvidia-cusparse-cu12==12.3.1.170
nvidia-cusparselt-cu12==0.6.2
nvidia-ml-py==12.570.86
nvidia-nccl-cu12==2.21.5
nvidia-nvjitlink-cu12==12.4.127
nvidia-nvtx-cu12==12.4.127
opencv-python==4.11.0.86
opt_einsum==3.4.0
packaging==24.2
parso==0.8.4
pexpect==4.9.0
pillow==11.1.0
prompt_toolkit==3.0.50
psutil==7.0.0
ptyprocess==0.7.0
pure_eval==0.2.3
Pygments==2.19.1
pynvml==12.0.0
pyparsing==3.2.1
pyperclip==1.9.0
python-dateutil==2.9.0.post0
pytorch-seed==0.2.0
pytorch_kinematics==0.7.4
PyYAML==6.0.2
requests==2.32.3
rich==13.9.4
sapien==3.0.0b1
scipy==1.15.2
shtab==1.7.1
six==1.17.0
smmap==5.0.2
stack-data==0.6.3
sympy==1.13.1
tabulate==0.9.0
toppra==0.6.0
torch==2.6.0
tqdm==4.67.1
traitlets==5.14.3
transforms3d==0.4.2
trimesh==4.6.2
triton==3.2.0
typeguard==4.4.2
typing_extensions==4.12.2
tyro==0.9.14
urllib3==2.3.0
wcwidth==0.2.13
```

StoneT2000, 2025-02-18:
This is likely due to Jax using most of the GPU memory by default, leaving none left for our renderer

Try running your script again but add the following environment variable

XLA_PYTHON_CLIENT_PREALLOCATE=false python script.py

nv-quan, 2025-02-18:
Thank you, it's working now. However, I hope there would be a more descriptive error message than "cannot create buffer" :sweat_smile:

StoneT2000, 2025-02-18:
We'll look into a nicer error message. We do this for camera instantiation actually but for window creation that is managed by a separate project

closed 2025-02-18 by StoneT2000