mani-skill/ManiSkill #853: [BUG] The truncated return keyword block the policy evaluation while using `gymnasium.vector.async_vector_env`

lyxichigoichie, 2025-02-18:
while I use maniskill task env (PickCube-v1) and `gymnasium.vector.async_vector_env` to evaluate my policy, I find that the output always reset every once in a while. Then I find that every time the `gymnasium.vector.async_vector_env` step the env, it would check the keywords `terminated` and `truncated`, look at [this](https://github.com/Farama-Foundation/Gymnasium/blob/main/gymnasium/vector/async_vector_env.py#L774). If `terminated==True` or `truncated==True`, the `gymnasium.vector.async_vector_env` will reset the env.

But maniskill set `truncated==True` while the global step is more than the task eposide max_step, look at [this](https://github.com/haosulab/ManiSkill/blob/main/mani_skill/utils/registration.py#L163), which means that if my policy can not finish the task in task eposide max_step then the env will reset. In fact, the step required by our policy is always more than the task eposide max_step since the task eposides are always obtained by motion planner and my policy is always shutdown in the halfway.

Now I can solve this bug by altering the [checking code](https://github.com/Farama-Foundation/Gymnasium/blob/main/gymnasium/vector/async_vector_env.py#L774) in gymnasium, but can we have any other methods to avoid the conflict？

Thanks!!  Best wishes

StoneT2000, 2025-02-18:
Usually the max episode steps is approximately tuned for RL

For IL generally you should set max episode steps of the environment to a higher number (I recommend 1.5x the mean length of demos in the dataset).

You can find some recommended setups in the eg diffusion policy baselines.

You also do not want to change any behavior with gymnasium, since if a policy doesn't solve the task you need to truncate the environment eventually.

lyxichigoichie, 2025-02-19:
> Usually the max episode steps is approximately tuned for RL
>
> For IL generally you should set max episode steps of the environment to a higher number (I recommend 1.5x the mean length of demos in the [...]

I find the key word and solve the problem, thanks!!

closed 2025-02-19 by lyxichigoichie