mani-skill/ManiSkill #855: Can't train the Diffusion Policy baseline with 'pointcloud', 'rgbd' observation modes ("TypeError: new(): data must be a sequence (got dict)")

jmorgan-bdai, 2025-02-18:
Hi,

I'm trying to train a diffusion policy model for `PickCube-v1` but am getting an error in `examples/baselines/diffusion_policy/train.py`.

``` bash
OBS_MODE="pointcloud"
RECORD_DIR=.maniskill/demonstrations__${OBS_MODE}
TASK="PickCube-v1"
CONTROL_MODE="pd_ee_delta_pose"
NUM_DEMOS=10

# Generate demonstrations
rm -rf $RECORD_DIR/$TASK/motionplanning/
python -m mani_skill.examples.motionplanning.panda.run \
    -e $TASK \
    --save-video \
    --obs-mode $OBS_MODE \
    --sim-backend "gpu" \
    --traj-name "trajectory" \
    --num-traj $NUM_DEMOS \
    --record-dir $RECORD_DIR   # runs headless and only saves video

# Convert the demonstrations to the desired control mode and visualize. Converting between joint space to pose needs to run on the cpu (see https://github.com/haosulab/ManiSkill/issues/850)
python -m mani_skill.trajectory.replay_trajectory \
  --traj-path $RECORD_DIR/$TASK/motionplanning/trajectory.h5 \
  --use-first-env-state \
  --target-control-mode $CONTROL_MODE \
  --obs-mode $OBS_MODE \
  --save-traj \
  --vis \
  --render_mode 'all' \
  --sim-backend "cpu"

# GPU training
python $BDAI/projects/maple/ws/src/external/maniskill/examples/baselines/diffusion_policy/train.py \
  --env-id $TASK \
  --demo-path $RECORD_DIR/$TASK/motionplanning/trajectory.$OBS_MODE.$CONTROL_MODE.physx_cpu.h5 \
  --total-iters 1000 \
  --control-mode $CONTROL_MODE \
  --sim-backend "gpu" \
  --num-eval-envs 3 \
  --no-capture-video
```

Error:
```

$ python path/to//maniskill/examples/baselines/diffusion_policy/train.py   --env-id $TASK   --demo-path $RECORD_DIR/$TASK/motionplanning/trajectory.$OBS_MODE.$CONTROL_MODE.physx_cpu.h5   --total-iters 1000   --control-mode $CONTROL_MODE   --sim-backend "cpu"   --num-eval-envs 3   --no-capture-video
/home/jstm/.local/lib/python3.10/site-packages/sapien/_vulkan_tricks.py:37: UserWarning: Failed to find Vulkan ICD file. This is probably due to an incorrect or partial installation of the NVIDIA driver. SAPIEN will attempt to provide an ICD file anyway but it may not work.
  warn(
Incompatible version of OpenGL_accelerate found, need at least (3, 1, 9) found (3, 1, 7)
/home/jstm/.local/lib/python3.10/site-packages/diffusers/utils/outputs.py:63: FutureWarning: `torch.utils._pytree._register_pytree_node` is deprecated. Please use `torch.utils._pytree.register_pytree_node` instead.
  torch.utils._pytree._register_pytree_node(
Incompatible version of OpenGL_accelerate found, need at least (3, 1, 9) found (3, 1, 7)
Incompatible version of OpenGL_accelerate found, need at least (3, 1, 9) found (3, 1, 7)
Incompatible version of OpenGL_accelerate found, need at least (3, 1, 9) found (3, 1, 7)
/home/jstm/.local/lib/python3.10/site-packages/diffusers/utils/outputs.py:63: FutureWarning: `torch.utils._pytree._register_pytree_node` is deprecated. Please use `torch.utils._pytree.register_pytree_node` instead.
  torch.utils._pytree._register_pytree_node(
/home/jstm/.local/lib/python3.10/site-packages/diffusers/utils/outputs.py:63: FutureWarning: `torch.utils._pytree._register_pytree_node` is deprecated. Please use `torch.utils._pytree.register_pytree_node` instead.
  torch.utils._pytree._register_pytree_node(
/home/jstm/.local/lib/python3.10/site-packages/diffusers/utils/outputs.py:63: FutureWarning: `torch.utils._pytree._register_pytree_node` is deprecated. Please use `torch.utils._pytree.register_pytree_node` instead.
  torch.utils._pytree._register_pytree_node(
Loading HDF5 file path/to/demonstrations__pointcloud/PickCube-v1/motionplanning/trajectory.pointcloud.pd_ee_delta_pose.physx_cpu.h5
Loaded
Load observations 10 <class 'dict'>
Load actions 10 <class 'numpy.ndarray'>
Traceback (most recent call last):
  File "path/to/maniskill/examples/baselines/diffusion_policy/train.py", line 326, in <module>
    dataset = SmallDemoDataset_DiffusionPolicy(args.demo_path, device, num_traj=args.num_demos)
  File "path/to/maniskill/examples/baselines/diffusion_policy/train.py", line 111, in __init__
    trajectories[k][i] = torch.Tensor(v[i]).to(device)
TypeError: new(): data must be a sequence (got dict)
Exception ignored in: <function AsyncVectorEnv.__del__ at 0x7f46706fa200>
Traceback (most recent call last):
  File "/home/jstm/.local/lib/python3.10/site-packages/gymnasium/vector/async_vector_env.py", line 549, in __del__
  File "/home/jstm/.local/lib/python3.10/site-packages/gymnasium/vector/vector_env.py", line 272, in close
  File "/home/jstm/.local/lib/python3.10/site-packages/gymnasium/vector/async_vector_env.py", line 465, in close_extras
AttributeError: 'NoneType' object has no attribute 'TimeoutError'
```

The pipeline works fine for with the `state` observation mode. 

Does the Diffusion Policy baseline only work with state input? If not, any idea how to fix this? 

Thanks!
- jeremy

chenyenru, 2025-02-19:
Not sure about point cloud, but for diffusion policy w/ rgbd there's an ongoing feature branch here: https://github.com/haosulab/ManiSkill/pull/643

StoneT2000, 2025-02-20:
@jmorgan-bdai We are merging in the diffusion policy with RGB input code somewhat soon. mostly just cleaning up code and verifying it solves some environments plus a cool drawing environment from rgb inputs. Also cleaning up our docs around how to run the IL baselines more.

jmorgan-bdai, 2025-02-21:
Ok awesome. Just subscribed to the pr so i'll give it a try once it's merged. Ya, docs could use a bit of work :) Mostly there though.

StoneT2000, 2025-02-22:
just heads up the PR was changed to this one #876 instead

StoneT2000, 2025-02-27:
closed now, newest code has rgb baselines (rgb+depth or depth only not tested)

closed 2025-02-27 by StoneT2000