mani-skill/ManiSkill #857: demonstration with dense reward

gxy-TJU, 2025-02-19:
I use the script  `motionplaning.sh` and `replay_for_il_baselines.sh` in `scripts/data_generation`, what I want  is a motion planning demonstrations with dense reward、 qpos、 and rgb. 
But I do not get it dirctly, they either lack rgb images or lack dense reward. 
Does there are one way I can get it directly? 
Thanks for any response.

StoneT2000, 2025-02-19:
Can you clarify what do you mean by you don't get it directly?

Is rgb not in the replayed trajectory data? It should be there

gxy-TJU, 2025-02-20:
> Can you clarify what do you mean by you don't get it directly?
> 
> Is rgb not in the replayed trajectory data? It should be there
Sorry,  I don't seem to express myself clearly.
What I run is this,

[image]
And the hf data I get is this,

[image]
There seems to be no dense reward here.

And also I run `motionplaning.sh`, In this case what I do not get is the rgb image.
All data I want is a motion planning demonstrations with dense reward、 qpos、 and rgb. 
Does there any way I can get it directly.
Thank you.

StoneT2000, 2025-02-20:
try adding `--reward-mode="normalized_dense"`

agent/qpos gives you the current qpos of the robot (although that is not the action taken in sim to get there).

closed 2025-02-20 by StoneT2000