mani-skill/ManiSkill #858: please teach me how to change the view of camera render ?

ShouZhengAI, 2025-02-19:
please teach me how to change the view of camera render ?

StoneT2000, 2025-02-19:
Which camera render in particular? For prebuilt tasks you can always modify sensor configurations (which include cameras) as well as the human render camera configurations https://maniskill.readthedocs.io/en/latest/user_guide/concepts/sensors.html

StoneT2000, 2025-02-19:
https://maniskill.readthedocs.io/en/latest/user_guide/concepts/sensors.html follow this to see how to override camera configurations, including camera pose

That doc has a link to the API https://maniskill.readthedocs.io/en/latest/user_guide/reference/mani_skill.sensors.html#mani_skill.sensors.camera.CameraConfig for the CameraConfig and what options are overridable
```
gym.make(..., sensor_configs=dict(shader_pack=args.shader, base_camera=dict(pose=sapien.Pose(...))
```

ShouZhengAI, 2025-02-20:
@StoneT2000 thanks for your reply, and I meet some trouble again. I change the code as your saying, but it seems not work,

code 
```
from typing import Callable, List, Type
import sys
sys.path.append('/')
import gymnasium as gym
import numpy as np
from mani_skill.envs.sapien_env import BaseEnv
from mani_skill.utils import common, gym_utils,sapien_utils
from mani_skill.sensors.camera import CameraConfig
import argparse
import yaml
from scripts.maniskill_model import create_model, RoboticDiffusionTransformerModel
import torch
from collections import deque
from PIL import Image
import cv2
import sapien

def parse_args(args=None):
    parser = argparse.ArgumentParser()
    parser.add_argument("-e", "--env-id", type=str, default="PushCube-v1", help=f"Environment to run motion planning solver on. ")
    parser.add_argument("-o", "--obs-mode", type=str, default="rgb", help="Observation mode to use. Usually this is kept as 'none' as observations are not necesary to be stored, they can be replayed later via the mani_skill.trajectory.replay_trajectory script.")
    parser.add_argument("-n", "--num-traj", type=int, default=25, help="Number of trajectories to test.")
    parser.add_argument("--only-count-success", action="store_true", help="If true, generates trajectories until num_traj of them are successful and only saves the successful trajectories/videos")
    parser.add_argument("--reward-mode", type=str)
    parser.add_argument("-b", "--sim-backend", type=str, default="auto", help="Which simulation backend to use. Can be 'auto', 'cpu', 'gpu'")
    parser.add_argument("--render-mode", type=str, default="rgb_array", help="can be 'sensors' or 'rgb_array' which only affect what is saved to videos")
    parser.add_argument("--shader", default="default", type=str, help="Change shader used for rendering. Default is 'default' which is very fast. Can also be 'rt' for ray tracing and generating photo-realistic renders. Can also be 'rt-fast' for a faster but lower quality ray-traced renderer")
    parser.add_argument("--num-procs", type=int, default=1, help="Number of processes to use to help parallelize the trajectory replay process. This uses CPU multiprocessing and only works with the CPU simulation backend at the moment.")
    parser.add_argument("--pretrained_path", type=str, default="/home/asrock/quanbao/ShouZheng/RoboticsDiffusionTransformer/mp_rank_00_model_states.pt", help="Path to the pretrained model")
    parser.add_argument("--random_seed", type=int, default=0, help="Random seed for the environment.")
    return parser.parse_args()

import random
import os

# set cuda 
args = parse_args()
# set random seeds

env_id = args.env_id
env = gym.make(
    env_id,
    obs_mode=args.obs_mode,
    control_mode="pd_joint_pos",
    render_mode=args.render_mode,
    reward_mode="dense" if args.reward_mode is None else args.reward_mode,
    sensor_configs=dict(shader_pack=args.shader,base_camera = dict(pose =sapien.Pose(p=[0.0, 0.5, 0.0], q=[0.0, 0.0, 0.0, 0.0]))),
    human_render_camera_configs=dict(shader_pack=args.shader),
    viewer_camera_configs=dict(shader_pack=args.shader),
    sim_backend=args.sim_backend
)

env.reset()

# Render image from environment
img = env.render().squeeze(0).detach().cpu().numpy()

# Create directory for saving images
image_save_dir = "images"
os.makedirs(image_save_dir, exist_ok=True)

# Set image path
image_path = os.path.join(image_save_dir, f"{env_id}_image_{1}.png")

# Save the rendered image
cv2.imwrite(image_path, cv2.cvtColor(img, cv2.COLOR_RGB2BGR))
print(f"Image saved at: {image_path}")
```

result:(the photo seems no change (also default))
[image]

wish your reply sincerely

ShouZhengAI, 2025-02-20:
version:
sapien                   3.0.0b1
mani_skill               3.0.0b18

StoneT2000, 2025-02-20:
Oh i just notice the RDT authors make a mistake. They are using the wrong cameras to render images for evaluation. You are not supposed to use env.render() to get images for policy inference, the images are in the returned obs object when calling env.reset and env.step

if render_mode=“rgb_array” then env.render renders from the cameras used for people to view. To change that camera modify the human render camera configs instead.

ShouZhengAI, 2025-02-20:
@StoneT2000 thank you! and I solve the problem by ` human_render_camera_configs=dict(shader_pack=args.shader,pose =sapien_utils.look_at([-0.6, 0.7, 0.6], [0.0, 0.0, 0.35])) `. this change can change the pose of ` env.render() `
thanks for your link doc.

closed 2025-02-20 by StoneT2000

Mickeyyyang, 2025-03-14:
> [@StoneT2000](https://github.com/StoneT2000) thank you! and I solve the problem by `human_render_camera_configs=dict(shader_pack=args.shader,pose =sapien_utils.look_at([-0.6, 0.7, 0.6], [0.0, 0.0, 0.35]))`. this modification can change the pose of `env.render()` thanks for your link doc.

Can I fix the camera on the end of the robotic arm? How should I change the script to achieve this?

StoneT2000, 2025-03-14:
Don't change human render configs. That is only used for rendering videos, not policy observation

The docs linked in my earlier comment explain which configs you can change. All mounted cameras are usually part of the list of sensors, not human render cameras or the viewer.

StoneT2000, 2025-03-14:
https://maniskill.readthedocs.io/en/latest/user_guide/concepts/sensors.html

change sensor configs (sensors in maniskill = any sensor like a camera used for policy observations)