mani-skill/ManiSkill #862: Issues with PPO baseline for PickSingleYCB-v1 task

seungbinjoo, 2025-02-20:
I'm having issues with my current RL setup (which builds on top of the `ppo.py` baseline provided by ManiSkill) to solve the PickSingleYCB-v1 task. To find where my issues are coming from I tried just running the `ppo.py` baseline, but `success_once` metric plateaus at around 0.55 after 50M steps of training (which is lower than what is expected based on the published results in the RL baselines documentation). I couldn't find the exact training configuration used to produce the published PPO results for PickSingleYCB-v1 task in the `/baselines/ppo/baselines.sh` file. Is this available anywhere?

StoneT2000, 2025-02-20:
Actually related #859. Apparently there are some objects being loaded that are not possible to grasp at all (easily) like a flat plate, so we are removing those from the env and redoing the RL results (I will delete it from wandb soon after testing). I think the old YCB results were on the corrected version of the environment.

I am re-running the ycb baseline again and updating the state and rgb based results in baselines.sh file probably by tomorrow.

seungbinjoo, 2025-02-20:
Brilliant, thanks so much for your help!

seungbinjoo, 2025-02-21:
Might be wrong here but as a side note, to redownload the correct YCB dataset using:
`python -m mani_skill.utils.download_asset ycb`
the SHA-256 hash for the ycb data source in `/mani_skill/utils/assets/data.py` should be updated to the new one (eb6f30642c90203715c178f67bf2288887ef6e7d05a9f3f1e713efcf7c2a541c) to avoid the following error:
`OSError: Downloaded file's SHA-256 hash does not match record: https://huggingface.co/datasets/haosulab/ManiSkill2/resolve/main/data/mani_skill2_ycb.zip`

StoneT2000, 2025-02-21:
Fixed now in #865 thanks!

PR #863 (2025-02-25): [Feature] Update PPO baselines for pick single YCB and remove wandb-entity CLI arguments
Closes #862
closed 2025-02-23 by StoneT2000