mani-skill/ManiSkill #874: Question about partial_reset parameter in training

songyuc, 2025-02-22:
Hello ManiSkill Team,

I'm studying the training acceleration techniques in ManiSkill and noticed the `partial_reset` parameter in the SAC implementation.

Could you please clarify:

1. Does enabling `partial_reset=True` actually improve training speed in practice? If yes, what's the typical acceleration ratio observed in your benchmarks?

2. Are there any trade-offs when using partial resets (e.g., impact on value estimation, exploration efficiency)?

3. Does this parameter require corresponding adjustments to other hyperparameters like `gamma` or `bootstrap_at_done`?

4. For final evaluation, is it mandatory to set `partial_reset=False` to ensure fair assessment?

It is said that 20-30% sample utilization improvement, but I couldn't find concrete speed comparisons. Would appreciate any official data or recommendations about this parameter's usage.

Thank you for building this amazing platform!

StoneT2000, 2025-02-22:
1. For PPO yes it improves training speed usually. For SAC it's probably not so noticeable
2. Good question, I'm not aware of much research in this however
3. Not that i am aware of
4. We turn partial resets off following the explanations here: https://maniskill.readthedocs.io/en/latest/user_guide/reinforcement_learning/setup.html#evaluation. In short, this enables one to randomize objects (if the environment is designed that way) since in GPU sims typically you cannot swap object shapes between env resets (you must delete the whole env), so partial resets do not make sense. With partial resets off in evals you can still track the two kinds of success conditions typically recorded, success_once and success_at_end. note that if you train with partial resets on, usually your agent will learn to maximize the success_once measure, not success_at_end.

Where is 20-30% from?

songyuc, 2025-02-23:
@StoneT2000 Thank you for your prompt and informative response! Your insights regarding the effectiveness of `partial_reset` for SAC and the potential impact on evaluation metrics are highly valuable. I appreciate your clarification on the need to disable `partial_reset` during final evaluation to ensure fair assessment and object randomization. 

Regarding the 20-30% sample utilization improvement, I apologize for not having the specific source readily available. I will conduct further research and experiments to obtain more accurate data on this matter.

Thank you once again for your assistance and for developing this amazing platform!

closed 2025-02-23 by songyuc