mani-skill/ManiSkill #875: [Issue / Question] Unable to open GUI Viewer via "human" render mode

CreativeNick, 2025-02-22:
I'm unable to open the GUI and view training when setting `render_mode` to "human". I need to set `--no-capture-video` in order to set the render mode to human. Otherwise, the GUI viewer will open for a second and then crash immediately upon receiving this error:

```
Epoch: 1, global_step=0
Evaluating
Traceback (most recent call last):
  File "/home/creativenick/Desktop/SimToReal/bimanual-sapien/ppo_rgb.py", line 455, in <module>
    eval_obs, eval_rew, eval_terminations, eval_truncations, eval_infos = eval_envs.step(agent.get_action(eval_obs, deterministic=True))
  File "/home/creativenick/Desktop/SimToReal/bimanual-sapien/ManiSkill/mani_skill/vector/wrappers/gymnasium.py", line 111, in step
    obs, rew, terminations, truncations, infos = self._env.step(actions)
  File "/home/creativenick/Desktop/SimToReal/bimanual-sapien/ManiSkill/mani_skill/utils/wrappers/record.py", line 467, in step
    obs, rew, terminated, truncated, info = super().step(action)
  File "/home/creativenick/anaconda3/envs/ms_dev/lib/python3.9/site-packages/gymnasium/core.py", line 461, in step
    return self.env.step(action)
  File "/home/creativenick/anaconda3/envs/ms_dev/lib/python3.9/site-packages/gymnasium/core.py", line 522, in step
    observation, reward, terminated, truncated, info = self.env.step(action)
  File "/home/creativenick/Desktop/SimToReal/bimanual-sapien/ManiSkill/mani_skill/utils/registration.py", line 161, in step
    observation, reward, terminated, truncated, info = self.env.step(action)
  File "/home/creativenick/anaconda3/envs/ms_dev/lib/python3.9/site-packages/gymnasium/wrappers/order_enforcing.py", line 56, in step
    return self.env.step(action)
  File "/home/creativenick/Desktop/SimToReal/bimanual-sapien/ManiSkill/mani_skill/envs/sapien_env.py", line 959, in step
    obs = self.get_obs(info)
  File "/home/creativenick/Desktop/SimToReal/bimanual-sapien/ManiSkill/mani_skill/envs/sapien_env.py", line 530, in get_obs
    obs = self._get_obs_with_sensor_data(info)
  File "/home/creativenick/Desktop/SimToReal/bimanual-sapien/ManiSkill/mani_skill/envs/sapien_env.py", line 597, in _get_obs_with_sensor_data
    sensor_data=self._get_obs_sensor_data(apply_texture_transforms),
  File "/home/creativenick/Desktop/SimToReal/bimanual-sapien/ManiSkill/mani_skill/envs/sapien_env.py", line 569, in _get_obs_sensor_data
    self.scene.update_render()
  File "/home/creativenick/Desktop/SimToReal/bimanual-sapien/ManiSkill/mani_skill/envs/scene.py", line 376, in update_render
    self._sapien_update_render()
  File "/home/creativenick/Desktop/SimToReal/bimanual-sapien/ManiSkill/mani_skill/envs/scene.py", line 385, in _sapien_update_render
    self.render_system_group.update_render()
RuntimeError: Modifying a scene (add/remove object/camera) is not allowed after creating the batched render system.
```

However, when doing so, the GUI viewer simply doesn't show up after adding `--no-capture-video` into the command.

StoneT2000, 2025-02-22:
are you using the GPU sim with parallel rendering?

The PPO code is not designed to support opening a viewer while training. We might be able to add this functionality perhaps although I don't see a clear use-case

CreativeNick, 2025-02-23:
Ah I see, yeah I'm using the GPU sim with parallel rendering. Thanks!