mani-skill/ManiSkill #881: Question about Replay Buffer implementation in sac.py

songyuc, 2025-02-25:
Hi guys,

I'm a student learning reinforcement learning by reading the code in this repository. I'm very impressed by the project!

I have a question about the implementation of the `ReplayBuffer` class in `examples/baselines/sac/sac.py`. I noticed that it uses a circular array (with `self.pos` to track the position) to store experiences, instead of a queue data structure.

I'm wondering why a queue (e.g., `collections.deque` in Python) wasn't used. `collections.deque` is implemented with a doubly linked list of blocks, and it also provides efficient append and pop operations. It can also be used with a `maxlen` parameter to achieve the circular behavior.

Is there a specific reason for choosing the circular array implementation over a queue? Are there any performance considerations or other factors that I might be missing?

Here's the link to the relevant code: https://github.com/haosulab/ManiSkill/blob/main/examples/baselines/sac/sac.py#L168-L171

Thank you for your time and consideration. I'm eager to learn more about this project's design choices.

StoneT2000, 2025-02-25:
storing cuda torch tensors is faster this way. Python deque is designed for python primitives and cpu objects

StoneT2000, 2025-02-25:
that being said i have been wondering if there's a more efficient circular buffer in torch using smth like torch.roll

closed 2025-02-27 by StoneT2000