mani-skill/ManiSkill #882: [Question] Reproducing Diffusion Policy Performance in PushT-v1 environment

SieRaX, 2025-02-25:
Hi,

Before asking my question, I would like to thank you for providing a great benchmark.

My issue is that the baseline code for Diffusion Policy does not achieve the expected performance in the PushT-v1 environment. I would appreciate any suggestions for improvement.

To verify whether the baseline code works as intended, I ran the shell script [baseline.sh](https://github.com/haosulab/ManiSkill/blob/5fb50fdeca88f23f8315cc32200d3d67247802b8/examples/baselines/diffusion_policy/baselines.sh) while disabling other environments.

Before running this script, I downloaded the [demonstration data](https://maniskill.readthedocs.io/en/latest/user_guide/learning_from_demos/setup.html) and executed [this script](https://github.com/haosulab/ManiSkill/blob/5fb50fdeca88f23f8315cc32200d3d67247802b8/scripts/data_generation/replay_for_il_baselines.sh#L32) to generate state-based demonstration data.

The trained model exhibits poor success rates, ranging between 0.15 and 0.2, as shown in the image below.

[image]

However, according to the paper ["Responsive Noise-Relaying Diffusion Policy: Responsive and Efficient Visuomotor Control"](https://arxiv.org/abs/2502.12724) (page 8), the state-based diffusion policy in the PushT-v1 environment should achieve a success rate between 0.4 and 0.5.

I have carefully reviewed the parameters and detailed settings, and they appear to be identical in both the paper and this repository.

Since one of the contributors is an author of the paper, I assume there might be additional details required to reproduce the exact results.

Does this script produce the same results as reported in the paper? What would be the best approach to accurately reproduce the results?

Thank you.

StoneT2000, 2025-02-25:
@xiuyuan0216 is the author on the paper and knows the details better. I think he uses either a slightly different DP or may have made some other changes compared to the default one on maniskill.

xiuyuan0216, 2025-02-26:
@SieRaX 

Thank you for your interest in the ManiSkill3 benchmark and for taking the time to read our newly released paper!

Although we used our own implementation of Diffusion Policy (DP) instead of the one provided by ManiSkill3, I can still reproduce the performance reported in our paper using the ManiSkill3 DP code.

One key detail to pay attention to is the number of demonstrations used for training. If you run the provided script directly, it sets demos=100 by default, meaning only 100 demonstrations are used for DP training. However, the full dataset actually contains around 700 demonstrations. By training on the entire dataset, I can achieve over 40% success rate using the ManiSkill3 DP code.

Please feel free to reach out to me at x1yuan@ucsd.edu if you have any further questions!

ZhangHangTao, 2025-11-05:
> [@SieRaX](https://github.com/SieRaX)
> 
> Thank you for your interest in the ManiSkill3 benchmark and for taking the time to read our newly released paper!
> 
> Although we used our own implementation of Diffusion Policy (DP) instead of the one provided by ManiSkill3, I can still reproduce the performance reported in our paper using the ManiSkill3 DP code.
> 
> One key detail to pay attention to is the number of demonstrations used for training. If you run the provided script directly, it sets demos=100 by default, meaning only 100 demonstrations are used for DP training. However, the full dataset actually contains around 700 demonstrations. By training on the entire dataset, I can achieve over 40% success rate using the ManiSkill3 DP code.
> 
> Please feel free to reach out to me at [x1yuan@ucsd.edu](mailto:x1yuan@ucsd.edu) if you have any further questions!

Dear authors, thank you for your work. I’d like to ask about the RGB-based diffusion policy—its performance on tasks like PickCube and StackCube-v1 seems unusual. Based on the training logs (https://wandb.ai/stonet2000/ManiSkill/runs/oj2viw5q?nw=nwuserstonet2000
) and my own experiments, we’ve been essentially unable to train this simple task successfully, which is quite surprising to me. It appears others are encountering the same issue as well (https://github.com/haosulab/ManiSkill/issues/1004
).

66Lau, 2026-01-12:
> [@66Lau](https://github.com/66Lau) I notice that you are using state observation, and I can get more than 40% success rate with ManiSkill3 DP code and 700 demos. I believe you can try tuning the hyperparameters a little bit and train for more iterations.

@xiuyuan0216 Thank you for your reply!

I have opened a new issue in https://github.com/haosulab/ManiSkill/issues/1375 with diffusion policy & ACT training and evaluation curve, where I found the performance is jittery with 4k episodes training. 

And thank you for that information. I am trying to tune DP for better performance!