mani-skill/ManiSkill #887: Incorrect articulation mesh for GPU backend

shaido987, 2025-02-26:
Using the GPU backend will return an incorrect mesh for articulations when using `get_articulation_meshes`.  If I use a CPU backend then it looks correct. Any idea what could be the issue here?

```python
import trimesh
import gymnasium as gym
from mani_skill.envs.sapien_env import BaseEnv
from mani_skill.utils.geometry.trimesh_utils import get_articulation_meshes

sim_backend = 'gpu'

env: BaseEnv = gym.make(
    'Empty-v1',
    num_envs=1,
    sim_backend=sim_backend,
)
obs, info = env.reset()

if env.gpu_sim_enabled:
    env.scene._gpu_apply_all()
    env.scene.px.gpu_update_articulation_kinematics()
    env.scene._gpu_fetch_all()

trimesh.Scene(get_articulation_meshes(env.scene.get_all_articulations()[0])).show()
```

Mesh when using GPU:

[image]

Mesh when using CPU:

[image]

StoneT2000, 2025-02-26:
That function was ported from ManiSkill2 but not tested for the physx cuda backend. Poses stored in cpu of links/actors by sapien (which is what the function is accessing right now) are set to the initial pose of the actor/link builder or set to the identity pose.

For the GPU sim equivalent there's a function for articulations. env.unwrapped.agent.robot. get_first_collision_mesh, which returns the collision mesh of the first articulation managed by the Articulation object. You can check the code if you need any more advanced usage.

trimesh doesn't have a batched option (something that I am hoping has a solution for) so it just returns the trimesh object for the first robot.

shaido987, 2025-02-26:
Thank you very much for the answer, the `get_first_collision_mesh()` method appears to be working correctly!