mani-skill/ManiSkill #891: Setting drive target for GPU enabled simulation

abhihjoshi, 2025-02-27:
I am trying to set the drive target for each of the individual joints in my simulation. I would like to run heterogeneous simulation where each instance of a scene contains different assets that potentially have different joint drive targets. When I ran the simulation with > 1 environment, I was met with this error:

`NotImplementedError: Setting drive targets of individual joints is not implemented yet.`

I was curious what this might entail to implement this feature and if there is a given timeline for this. Also, I was curious if there was a workaround for this. Would appreciate any help on this matter!

StoneT2000, 2025-02-27:
Are you designing your own custom controller? If you use any of the default envs/robots those will likely override what you are trying to set

And currently it's simply not implemented, for gpu sim your code won't work, i can add this as a feature request to add later this week

StoneT2000, 2025-02-27:
Oh i may have misread, are these joint targets for various articulations and not the maniskill managed robot articulations?

abhihjoshi, 2025-02-27:
Thanks for the quick response! Yes, the joint articulations are not for the robot. They are for other objects in the scene.

abhihjoshi, 2025-02-27:
Also, as a follow up, if this logic is added in _load_scene for other non-robot articulations, would this be considered valid?

StoneT2000, 2025-02-27:
it probably won't work since load scene occurs before the GPU sim is even initialized. You can do it in initialize episode though if you want to set something once per episode?

abhihjoshi, 2025-02-27:
I think I might be misunderstanding how heterogeneous simulation works. From what I understand, when we want to create many parallel environments, `_load_scene` is called once at the beginning to load the assets and `_load_episode` is called every time we start a new episode where we can do some domain randomization. Assuming I don't care about domain randomization, would adding this information in `_load_scene` be equivalent to adding in `_load_episode`?

StoneT2000, 2025-02-27:
does `_load_episode` mean `_initialize_episode` here? (we only have init episode, no load episode function.

The process for env resets is always
1. reconfigure (load_scene, called only once)
2. initialize episode 

Some things cannot be randomized in initialize episode like object geometry. Other things can like drive targets, object colors, joint stiffness/damping values.

What are you trying to randomize across parallel envs / heterogeneously simulate?

abhihjoshi, 2025-02-28:
I am trying to randomize similar to the OpenCabinet environment with a different asset for each sub environment. Following that environment, I added the set drive logic in the _load_scene function.

StoneT2000, 2025-03-11:
That won't work or at least it shouldn't in GPU sim. drive targets determine where the joint will try to move towards at each timestep, and I think is only available after the scene is loaded and sim is initialized for physx cuda backend.

Are you trying to randomize the initial qpos of the assets instead?

closed 2025-10-21 by StoneT2000