mani-skill/ManiSkill #892: The issue about trajectory replay.

YZY14606, 2025-02-27:
I used the trajectory replay script `mani_skill.trajectory.replay_trajectory` in ManiSkill3 to replay the .h5 and .json files we collected. However, I found that the replayed trajectory always differs from the trajectory generated during the initial recording.

For example, in our object pushing task, the original trajectory of the robotic arm does not collide with the object in the vertical direction. However, in the replayed trajectory, the robotic arm collides with the object in the vertical direction.

I’m not sure that the trajectory data we collected indicates that the robotic arm will collide with the object in the vertical direction?

At the same time, during trajectory reproduction, the simulator always throws the following error:
```
Episode 0 is not replayed successfully. Skipping                                
Episode 1 is not replayed successfully. Skipping                                
Episode 2 is not replayed successfully. Skipping                                
Episode 3 is not replayed successfully. Skipping                                
Episode 4 is not replayed successfully. Skipping                                
Episode 5 is not replayed successfully. Skipping                                
Episode 6 is not replayed successfully. Skipping                                
Episode 7 is not replayed successfully. Skipping                                
Episode 8 is not replayed successfully. Skipping                                
Episode 9 is not replayed successfully. Skipping
```
I'm not sure whether this is caused by the trajectory replay tool or the data we collected？

YZY14606, 2025-02-27:
I have resolved the issue with the accuracy of trajectory replay by using `--use-env-states`. However, I have not been able to find a solution to the above error.What could be the possible reason for this?

StoneT2000, 2025-02-27:
What script did you run? Your own trajectory data or the public ones on hugging face?

YZY14606, 2025-02-28:
It's my own trajectory data. The data are as follows:

[20250228_100909.zip](https://github.com/user-attachments/files/19019438/20250228_100909.zip)

YZY14606, 2025-03-02:
At the same time, when I use the following code to save the video, a save failure error occurs, and I am unable to obtain the video.
`python -m mani_skill.trajectory.replay_trajectory --use-env-states --traj-path /home/yzy/nus/robot/data_generation/demos/PushCup-yzy/motionplanning/20250228_100909.h5 --save-video`.

Error:
`Replayed 100 episodes, 0/100=0.00% demos saved.`

StoneT2000, 2025-03-02:
Do you have the code for the environment you made? I will need that to reproduce the error

YZY14606, 2025-03-03:
[env_creation.zip](https://github.com/user-attachments/files/19046880/env_creation.zip)
This is my env.

StoneT2000, 2025-03-04:
https://github.com/user-attachments/assets/94332fc1-181b-4a76-9256-99e6ab1eacfa

I dont have the bowl assets so i swapped them with a white cube. The video above is what the first demo looks like.

Your environment success condition checks if the object is placed above the target, but it looks like your demos all don't solve the task (or even open the gripper).

If you want to try and debug further add --allow-failure along with --save-video. It will then save the videos even if it fails.

YZY14606, 2025-03-05:
I'm sorry for the delayed response. I've been researching this issue over the past few days. The action mode in my script is pd joint pos and I haven't changed it.

I think the reason why the reproduced video doesn't match the original one is due to the random seed. I performed random initialization of the poses for both the Panda robotic arm and the object at the start of each trajectory.
After experimenting, I found that when I write my own code to initialize the poses, sizes, and the robot’s pose using the recorded initial object poses, sizes and the robot pose, and then use the recorded actions for each step, the video matches the original one.
At this point, the system only encounters the error: `Episode is not replayed successfully. Skipping.`

StoneT2000, 2025-03-06:
Ah I know what you mean. The environment states do not encode object geometry like object scale, it only ensures the poses are the same.

Currently in your code what exactly is being randomized at scene loading stage and episode initialization? Generally object geometry should and only ever be randomized in def _load_scene and when doing so you should use the `self._batched_episode_rng` object (numpy rng but batched) to ensure better reproducibility as well as reproducibility on the level object geometry randomization between CPU and GPU sim when resetting with the same seeds.

YZY14606, 2025-03-06:
Okay! Thanks for your detailed replay.

closed 2025-03-06 by StoneT2000

abadithela, 2025-04-15:
Hi @StoneT2000 

I am having some issues with reproducibility in GPU parallelized environments. For testing purposes, I am evaluating a policy on the same initial condition 30 times in parallel. How do I use _batched_episode_rng? Since it is a private variable, I assumed that the function call to reset with a seed list should resolve the randomness across parallel environments:

```
def reset(self, seed: Union[None, int, list[int]] = None, options: Union[None, dict] = None):
        """Reset the ManiSkill environment with given seed(s) and options. Typically seed is either None (for unseeded reset) or an int (seeded reset).
        For GPU parallelized environments you can also pass a list of seeds for each parallel environment to seed each one separately.

        If options["env_idx"] is given, will only reset the selected parallel environments. If
        options["reconfigure"] is True, will call self._reconfigure() which deletes the entire physx scene and reconstructs everything.
        Users building custom tasks generally do not need to override this function.

        Returns the first observation and a info dictionary. The info dictionary is of type

        .. highlight:: python
        .. code-block:: python

            {
                "reconfigure": bool # (True if the env reconfigured. False otherwise)
            }

        Note that ManiSkill always holds two RNG states, a main RNG, and an episode RNG. The main RNG is used purely to sample an episode seed which
        helps with reproducibility of episodes and is for internal use only. The episode RNG is used by the environment/task itself to
        e.g. randomize object positions, randomize assets etc. Episode RNG is accessible by using `self._batched_episode_rng` which is numpy based and `torch.rand`
        which can be used to generate random data on the GPU directly and is seeded. Note that it is recommended to use `self._batched_episode_rng`
        if you need to ensure during reconfiguration the same objects are loaded. Reproducibility and seeding when there is GPU and CPU simulation can be tricky and we recommend reading
        the documentation for more recommendations and details on RNG https://maniskill.readthedocs.io/en/latest/user_guide/concepts/rng.html

        Upon environment creation via gym.make, the main RNG is set with fixed seeds of 2022 to 2022 + num_envs - 1 (seed is just 2022 if there is only one environment)
        During each reset call, if seed is None, main RNG is unchanged and an episode seed is sampled from the main RNG to create the episode RNG.
        If seed is not None, main RNG is set to that seed and the episode seed is also set to that seed. This design means the main RNG determines
        the episode RNG deterministically.

        """
```

So, I am calling the reset function to each parallel environment with the same seed:

`obs, info = env.reset(options=eval_env_config, seed=[cfg.seed for k in range(env_cfg.num_envs)])`

However, this still results in different evaluation outcomes for the same policy. I have made sure that the policy is deterministic.

StoneT2000, 2025-04-16:
_batched_episode_rng is slower typically (its a literal list of numpy RNG objects). It is used typically for reconfiguration time (def _load_scene and def _load_agent) (doesn't always occur during a reset) randomizations, which means object geometries, sizes, textures etc. usually (depends on how the task is written).

However the task initialization (def _initialize_episode) always occurs each env reset, so it needs to be relatively faster, so _batched_episode_rng is not used and instead torch random is typically used to generate batched numbers faster.

If you really want the initial state to be the same use env.set_state_dict(some_state_dict). State dict is retrieved via env.get_state_dict() and will be batched in dict leaves by number of parallel environments. This state setting method also lets you run things mostly deterministically in GPU sim even with a different number of parallel environments than the one used to get the state dict originally. This strategy is used for trajectory replay, as torch RNG is not element wise reproducible (so a different number of parallel envs will always yield different initial states).