mani-skill/ManiSkill #900: Every time install maniskill using pip, it will automatically install the newest version of torch which covers the torch in local env

lyxichigoichie, 2025-03-03:
Hi,
every time I install maniskill using pip, it will automatically download and install the newest version of torch which covers the torch in local env. I remeber that we can select the torch that is compatible with our system saying in the documentation.

I check the dependencies in `setup.py` and find that the [pytorch_kinematics package](https://github.com/haosulab/ManiSkill/blob/main/setup.py#L44) set torch as its dependencies([link](https://github.com/Silverster98/pytorch_kinematics/blob/main/setup.py#L13)) which causes the automatical download.

Could you please fix this problem? It is unconvenient for me since I have to specify the version of torch for satifying the need of some packages that need compliling like pytorch3d.

Thanks.

StoneT2000, 2025-03-04:
I see, I'll see if we can install pytorch kinematics without dependencies and manage them ourselves.

StoneT2000, 2025-03-13:
Ok it seems like this is not possible actually.

The solution here when installing mani_skill to avoid accidentally installing a later torch version is to run

```
pip install mani_skill torch==<your_current_version>
```

closed 2025-03-13 by StoneT2000