mani-skill/ManiSkill #901: Spinning robot gripper during initialization

abhihjoshi, 2025-03-03:
I am currently trying to initialize a floating Panda gripper in my environment, however, I notice that the gripper occasionally spins at the beginning when running the environment. When I render the env, the gripper is not colliding with any other objects in the scene. I am uncertain if this is because the gripper is colliding with itself (not sure how to check this). I was curious what could be the cause for this? Any help is appreciated!

StoneT2000, 2025-03-04:
Any example code? is this your own floating panda gripper urdf?

StoneT2000, 2025-03-06:
This code was used to apply any pose changes to cabinet/articulations, then update the articulation kinematics and re-fetch all the cabinet data (qpos, link poses) on the GPU. We need the latest link poses (computed based on the current qpos and articulation root pose, which is only updated when you call apply all and update art kinematics) to compute handle poses for generating goal information.

Unfortunately I forgot why we take an extra physics step here however, it might be due to a hack to try and fix some other bug a while back but might not be needed anymore

ReflectiveOwl, 2025-03-06:
I also have a similar issue with the Kinova arm + Robotiq gripper. I have a similar problem to what was reported in [this issue](https://github.com/haosulab/ManiSkill/issues/477). When I put the Robotiq 2F-140 gripper together with the Kinova Gen 3 robot, the arm starts to spin like crazy, and the actions on the gripper affect the whole body of the robot. I tried to tune the PD controller, to calculate arrays in _after_loading_articulation as in the CVPR tutorial and many more things, but it didn't help

Decreasing force limits on gripper joints from 100 to less than 0.1 values helps to stabilize it, but the gripper doesn't work then.

[robot_spinning.mp4](https://github.com/user-attachments/assets/dab70a14-3bae-48ce-986d-503fc57743e8)

[code_and_urdf.zip](https://github.com/user-attachments/files/19111185/code_and_urdf.zip)

StoneT2000, 2025-03-13:
Thanks for the inputs @abhihjoshi @ReflectiveOwl 

Let me first take a look at the panda one. Can @abhihjoshi you share your URDF and agent class code?

ReflectiveOwl, 2025-03-21:
After some debugging, I was able to fix gripper instability when tested on random actions or keyframe actions as in the [manual](https://maniskill.readthedocs.io/en/latest/user_guide/tutorials/custom_robots.html). 

But I got a very strange behavior when trying to grasp an object. The gripper can hold the object, but then the whole body becomes unstable and starts to rotate. I tried to measure contact forces similar to how it was implemented for the default panda robot, but both l_contact_forces and r_contact_forces vectors are [0, 0, 0] at any time (for both outer knuckles and inner fingers), even when the object is grasped. It looks like the issue is with the gripper, but I'm not sure how to fix it. There's also some problem with the stability of the whole robot if I use .srdf as generated using the official Kinova repository, but disable_self_collisions = True fixes it for now.

Here is the video, code, and command to reproduce the issue:
python test_kinova.py -r "kinova" -c "pd_joint_delta_pos" --zero-actions

[code_and_urdf.zip](https://github.com/user-attachments/files/19390178/custom_demo_robot.zip)

https://github.com/user-attachments/assets/8bbe7533-1fde-48fc-99ee-a9cda162b84e

Any help would be appreciated

StoneT2000, 2025-04-03:
Sorry just got back to this issue @abhihjoshi 

Let me resolve that first.

Running

```
python mani_skill/examples/demo_robot.py -r floating_panda_gripper --zero-actions -b gpu -c pd_joint_delta_pos
```

seems to be fine on my end. This is our implemented floating gripper. Were you finding this issue only when you took that extra physx step after updating kinematics?

closed 2025-05-20 by StoneT2000