mani-skill/ManiSkill #902: RuntimeError: vk::Device::createFenceUnique: ErrorOutOfHostMemory

ZhefeiGong, 2025-03-04:
Hi, thanks for your great work on ManiSkill!

I'm currently facing a serious `RuntimeError: vk::Device::createFenceUnique: ErrorOutOfHostMemory` issue. Any help would be greatly appreciated!

## setting
* `mani_skill`-`3.0.0b1`
* `ubuntu20.04`

## phenomenon
I'm using the ManiSkill environment for `PegInsertionSide-v1` using `AsyncVectorEnv` to run `image-based` (`RGB`) experiments in `CPU` parallel. After thousands of steps, it crashes due to CPU limitations. I've tested CPUs with 16, 32, and 64 cores, and the environment consistently fails at around 50k, 100k, and 250k steps, respectively.

I also tried changing `obs_mode` to `state`, which works fine. Given the previous issue you resolved [here](https://github.com/haosulab/ManiSkill/issues/209) and the error message, I suspect there might be a similar `fence leak`, but I'm not sure how to fix it. Could you take a look? Thanks!

## Approximate code
```
def make_env():
    def thunk():
        # env initialization
        env = gym.make('PegInsertionSide-v1', 
                       reward_mode='sparse', 
                       obs_mode='rgb', 
                       control_mode='pd_ee_delta_pose',
                       render_mode=None,
                       max_episode_steps=300)
        # env wrap
        if video_dir: env = RecordEpisode(env, output_dir=video_dir, save_trajectory=False, info_on_video=True)
        env = SelfRecordEpisodeStatistics(env) # self-design
        env = gym.wrappers.ClipAction(env)
        env = SelfSeqObservationWrapper(env, 2) # self-design
        env = SelfSeqActionWrapper(env) # self-design
        # env seed
        env.action_space.seed(seed)
        env.observation_space.seed(seed)
        return env
    return thunk

VecEnv = gym.vector.SyncVectorEnv if args.sync_venv or args.num_envs == 1 \
    else lambda x: gym.vector.AsyncVectorEnv(x, context='forkserver')
envs = VecEnv(
    [make_env() for i in range(args.num_envs)]
)
```

## Error
```
[730] Traceback (most recent call last):
[731]   File "***", line 192, in <module>
[732]     next_obs_seq, rewards, terminations, truncations, infos = envs.step(final_act_seq)
[733]   File "***/lib/python3.10/site-packages/gymnasium/vector/vector_env.py", line 204, in step
[734]     return self.step_wait()
[735]   File "***/lib/python3.10/site-packages/gymnasium/vector/async_vector_env.py", line 333, in step_wait
[736]     self._raise_if_errors(successes)
[737]   File "***/lib/python3.10/site-packages/gymnasium/vector/async_vector_env.py", line 555, in _raise_if_errors
[738]     raise exctype(value)
[739] RuntimeError: vk::Device::createFenceUnique: ErrorOutOfHostMemory
```

StoneT2000, 2025-03-04:
Can you try running the environment code, just roll it out, without any other code or extra wrappers (keep the CPUGymWrapper if possible).

Moreover can you upgrade maniskill to b19. We fixed a bug with the CPU sim renderer that was causing it to run very slow and take up extra GPU memory on cuda enabled gpus.

ZhefeiGong, 2025-03-11:
Thanks for your prompt response! 
I've updated the version, and it seems to be working well now—no more crashes. 
However, I’ve noticed that reinforcement learning training on `vision-based` tasks is still quite slow. I’ve also tried using the `GPU` simulator but haven’t seen significant improvements. Do you have any suggestions for speeding up the simulation, such as built-in support for a `multi-GPU` setup?

StoneT2000, 2025-03-11:
How slow is slow, which tasks are you are testing?

ZhefeiGong, 2025-03-12:
Hi Stone,
I'm testing the `PegInsertionSide-v1` and `PlugCharger-v1`. 
Oh, unfortunately, the crash emerged again. 
This time failed after 500k steps on `image-based (RGB)` in `CPU` parallel. I've already updated to the newest version. 
Everything works fine on `state-based` in `CPU` parallel
Any ideas on how to solve it?

```
[8932] Traceback (most recent call last):
[8933]   File "******", line 192, in <module>
[8934]     next_obs_seq, rewards, terminations, truncations, infos = envs.step(final_act_seq)
[8935]   File "******/lib/python3.10/site-packages/gymnasium/vector/vector_env.py", line 204, in step
[8936]     return self.step_wait()
[8937]   File "******/lib/python3.10/site-packages/gymnasium/vector/async_vector_env.py", line 333, in step_wait
[8938]     self._raise_if_errors(successes)
[8939]   File "******/lib/python3.10/site-packages/gymnasium/vector/async_vector_env.py", line 555, in _raise_if_errors
[8940]     raise exctype(value)
[8941] RuntimeError: vk::Device::createFenceUnique: ErrorOutOfHostMemory

```

ZhefeiGong, 2025-03-12:
16 parallel environments, and we use A100 GPU which has around 80G memory

StoneT2000, 2025-03-12:
Can you try 
- running with just 4 parallel envs
- run with 16 but remove all wrappers and simply take random steps and see how long it lasts (remove the self… wrappers, record episode etc)

ZhefeiGong, 2025-03-12:
Great, I'll have a try

closed 2025-04-09 by ZhefeiGong

ZhefeiGong, 2025-04-10:
Hi Stone, thanks again for your helpful guidance earlier!

Unfortunately, I wasn't able to resolve the issue — I'm still encountering the `RuntimeError: vk::Device::createFenceUnique: ErrorOutOfHostMemory` error when using `obs_mode = image`, even after trying your suggestions. I suspected it might be related to a memory (fence) leak, but I wasn’t able to pinpoint the exact reason at the time. Then, I switched to a different environment due to my work schedule.

I really appreciate your support, and I’ll definitely keep a close eye on this issue moving forward.

StoneT2000, 2025-04-10:
Sure thing. I have not been able to reproduce this exactly yet (maybe try using the mani-skill-nightly package), but I'll keep this issue in mind if someone else also gets it.