mani-skill/ManiSkill #905: Update Articulation/Actor builders to create different physical/render materials if requested to enable easier modification of object in each sub-scene individually.

StoneT2000, 2025-03-06:


shaido987, 2025-03-06:
This should make code for adding various domain randomizations much cleaner. Would robot agents also be included in this?

StoneT2000, 2025-03-06:
Yeah this is one of the original reasons for the issue.

Currently we have a rather complex  system for loading controllable robots/articulations primarily because robot/controller code is complex and we wanted to abstract that away from users. As a result it's kind of inflexible when it comes to robot configurations like DR and modifying materials (and if you want to DR physical / visual properties the code requires more knowledge of how sapien works which is not the expectation for maniskill users).

Fixing up some new changes to the robot class API probably (or more like improve existing docs and give more example code) and the actor builder/articulation builder APIs to make it a bit easier.

closed 2025-03-07 by StoneT2000

StoneT2000, 2025-03-07:
https://maniskill.readthedocs.io/en/latest/user_guide/tutorials/domain_randomization.html#agent-robot-and-controller-randomizations updated docs on how to randomize robot properties

PR #910 (2025-03-07): [Feature] update documentation on actor/link domain randomization and make it possible to domain randomize robots/agents
You can now domain randomize robots/agents provided in the custom environment you define

```
def _load_agent(self, options: dict):
    # in addition to setting agent initial poses you can turn on the option to build each agent separately and merge them which enables per-environment randomizations
    # of all physical and visual properties
    super()._load_agent(options, initial_agent_poses=sapien.Pose(), build_separate=True)
```
passing `build_separate=True` will ensure a [... cut]