mani-skill/ManiSkill #913: draw something with motionplanning

gxy-TJU, 2025-03-07:
I want to drow something with pandastick like your demo,

https://github.com/user-attachments/assets/35028db2-2187-4dc0-96cb-93d8b42a97e2

May be a word 'B' or else, I can precompute the coordinate of the points I need to draw, and I want to konw how to control my robot with tcp pose?
Are there any demo I can study?

StoneT2000, 2025-03-07:
We have some motion planning solutions used to generate the triangle drawing one. You can look at that code as a reference for how to write solutions for other shapes https://github.com/haosulab/ManiSkill/blob/main/mani_skill/examples/motionplanning/panda/solutions/draw_triangle.py

arnavg115, 2025-03-08:
I'd like to add that if you have a set of points you can convert to an SVG you may be able to use the DrawSVG environment and input your own custom svg paths as an argument to the environment. Otherwise the motion planning solution for DrawSVG also uses a list of points to draw shapes so it may be a helpful reference. https://github.com/haosulab/ManiSkill/blob/main/mani_skill/examples/motionplanning/panda/solutions/draw_svg.py

gxy-TJU, 2025-03-10:
Thank your response! I have made it!

closed 2025-03-10 by StoneT2000