mani-skill/ManiSkill #914: The issue of camera params.

YZY14606, 2025-03-07:
I recently encountered some issues when using the camera parameters.

My Base_camera pose is set as follows: `pose = sapien_utils.look_at(eye=[0, 0, 0.4], target=[0, 0, 0])`.

However, when I use the `get_params()` function to retrieve the camera's extrinsic parameters, the result does not match the actual camera pose. They are as follows:

extrinsic_cv:
```
 [[ 0.0000000e+00 -1.0000000e+00  0.0000000e+00  0.0000000e+00]
 [-9.9999994e-01  0.0000000e+00 -5.9604645e-08  0.0000000e+00]
 [ 5.9604645e-08  0.0000000e+00 -9.9999994e-01  2.0000000e-01]]
```
cam2world_gl:
```
[[ 5.9604645e-08  9.9999994e-01  0.0000000e+00  0.0000000e+00]
 [-9.9999994e-01  5.9604645e-08  0.0000000e+00  0.0000000e+00]
 [ 0.0000000e+00  0.0000000e+00  1.0000000e+00  4.0000001e-01]
 [ 0.0000000e+00  0.0000000e+00  0.0000000e+00  1.0000000e+00]]
```
It can be seen that both the extrinsic_cv and the cam2world_gl do not match the actual pose of the camera. 

The matrix of the camera relative to the world coordinate system should be：
```
[1,0,0 ,0]                   [-1,0,0,0]
[0,-1,0, 0]    or           [0,1,0,0]
[0,0,-1, 0.4]               [0,0,-1,0.4]
```
I can't understand the function's results. And I also have concerns about the obtained intrinsic_cv.

StoneT2000, 2025-03-07:
extrinsic_cv is the extrinsic matrix. cam2world_gl is the model matrix. These transforms transform things differently

The code for the extrinsic in GPU and CPU sim are the same. you take the pose of the camera in the global frame, take inverse (and for convention we use opencv convention and right matrix multiply to swap rows).

```
ros2opencv = torch.tensor(
                [[0, 0, 1, 0], [-1, 0, 0, 0], [0, -1, 0, 0], [0, 0, 0, 1]],
                device=self.scene.device,
                dtype=torch.float32,
            ).T
            res = (
                ros2opencv @ self.get_global_pose().inv().to_transformation_matrix()
            )[:, :3, :4]
```

cam2world_gl is useful for e.g. processing pointcloud data. In our code we do something like

```
cam2world = camera_params[cam_uid]["cam2world_gl"].to(position.device)
            xyzw = torch.cat([position, segmentation != 0], dim=-1).reshape(
                position.shape[0], -1, 4
            ) @ cam2world.transpose(1, 2)
```
to get the pointcloud in world coordinates from camera coordinates (where depth info is a distance to the camera, not 0)

StoneT2000, 2025-03-07:
re the rotations looking different. They are pretty close to 1s and 0s, probably some floating point precision error. For the columns being swapped this is just some convention stemming from SAPIEN's original convention for these matrices. In fact our point cloud code will actually just transpose row 1 and 2 for that reason.

YZY14606, 2025-03-08:
Thanks for your response! I would like to ask if there is a function in ManiSkill3 that can directly convert points from the pixel coordinate system to the world coordinate system？

StoneT2000, 2025-03-09:
https://github.com/haosulab/ManiSkill/blob/7b71cedb92367ae294a130dc08a504e6ed530fa7/mani_skill/envs/utils/observations/observations.py copy the sensor data to pointcloud function here.

YZY14606, 2025-03-10:
Thanks very much!

closed 2025-03-10 by YZY14606