mani-skill/ManiSkill #920: Does the motion planner have a significant error?

YZY14606, 2025-03-10:
I used the script  for motion planning from `mani_skill/examples/motionplanning/panda/motionplanner.py`. However, when I used it, I found that this motion planner seemed to have some errors.
I ran the following code：
```
    res = planner.move_to_pose_with_screw(goal_pose)
    print(goal_pose)
    print(env.agent.robot.links_map["panda_hand_tcp"].pose)
```
The results are as follows:
```
Pose([0.036, -0.008, 0.022], [-0.00253994, 0.805884, 0.592067, 0.00100953])
Pose(raw_pose=tensor([[ 3.1570e-02, -1.0900e-02,  2.2202e-02, -3.2795e-03,  8.0596e-01,
          5.9196e-01,  6.3060e-04]]))
```
It can be observed that there is some discrepancy between our goal_pose and the pose of panda_hand_tcp. Sometimes, this may lead to the failure of our planned target.
I would like to ask whether this error is caused by the motion planner？Or is it that this goal_pose is not meant for panda_hand_tcp?

StoneT2000, 2025-03-11:
You can try playing with mp-lib further to tune this. The motionplanner is not heavily tuned to be extremely precise or accurate, just a tool to use and modify. You can probably try running move to pose with screw more than once, another is to lower the joint vel/acc limits of the planner so it is less likely to overshoot.

StoneT2000, 2025-03-11:
@lexseal (mplib maintainer) might have a better idea

YZY14606, 2025-03-12:
Okay, thank you for your response.

closed 2025-04-27 by StoneT2000