mani-skill/ManiSkill #921: The query about obs-mode setting.

YZY14606, 2025-03-10:
I’m very sorry to bother you.

I would like to ask: When we set `obs-mode` to `none`, then we can‘t obtain camera data using `camera.capture()` and `camera_obs = camera.get_obs()`? I want to set `obs-mode` to `none`, and the simulator will run faster.

When we set `obs-mode` to `segmentation`, I found that we can still obtain the camera's position information. I'm not sure whether this usage is correct？

YZY14606, 2025-03-10:
Here is the latest station. When I set `obs-mode` to `none`, I found that sometimes the camera can successfully capture image data, while other times it cannot. I'm not sure what the reason is.

StoneT2000, 2025-03-11:
obs-mode only controls what the returned observation of `env.reset` and `env.step` contain.

I don't quite follow what your objective is though? Do you want to generate videos but don't generate observations?

YZY14606, 2025-03-12:
I want to manually record a segmentation image from the camera while setting `obs-mode` to `none`, as this can make my program run faster.

However, I found that when `obs-mode` is set to `none`, the camera cannot obtain the `object segmentation IDs`. Only when I add the `--save-video` command does the camera function successfully.

I would like to ask: When `obs-mode` is set to `none` and the `--save-video` command is not used, does that mean we cannot obtain the correct camera image data?

I'm in CPU simulation.

StoneT2000, 2025-03-13:
Cameras need to be initialized or else the IDs do not exist. 

A lazy way to initialize maybe is to call env.render_sensors() once. Then use the internal functions to get segmentation images accordingly.

Otherwise I would also just recommend setting obs-mode to "segmentation", we then auto generate the seg images at each env step

YZY14606, 2025-03-17:
In this image, after using gym.make, `env.reset()` is automatically called.
[image]

At the same time, I also tried manually using the `env.reset()` function. When I used it after gym.make:
```
    env = gym.make(
        env_id,
        obs_mode=args.obs_mode,
        control_mode="pd_joint_pos",
        render_mode=args.render_mode,
        reward_mode="none" if args.reward_mode is None else args.reward_mode,
        sensor_configs=dict(shader_pack=args.shader),
        human_render_camera_configs=dict(shader_pack=args.shader),
        viewer_camera_configs=dict(shader_pack=args.shader),
        sim_backend=args.sim_backend,
        num_envs = 1,
        reconfiguration_freq = 1
    )
    env.reset()
    env.render_sensors()
```
There will be error:
```
  File "/home/yzy/miniconda3/envs/nus_push_net/lib/python3.9/site-packages/mani_skill/utils/visualization/misc.py", line 63, in tile_images
    if len(images[0].shape) == 4:
IndexError: list index out of range
```

When I use `self.reset()` and `self.render_sensors()` in the `_load_scene`, the program will keep running indefinitely, as if it has frozen.

StoneT2000, 2025-03-17:
Ah okay I know where the issue is.

When observation mode is None we will always default to capturing no data with sensors through any of the public apis. render_sensors is meant to render what the sensors are capturing, which is set to none.

A work around is to use the direct camera APIs to update the renderer, capture an image, and then fetch whatever modalities you want.

```python
import gymnasium as gym
import mani_skill.envs
if __name__ == "__main__":
    env = gym.make("PushCube-v1",  obs_mode="none")
    obs, info = env.reset()
    base_env = env.unwrapped
    base_env.scene.update_render(update_sensors=True, update_human_render_cameras=False)
    base_env.scene.sensors["base_camera"].capture()
    data = base_env.scene.sensors["base_camera"].get_obs(rgb=True, depth=True, segmentation=True, normal=False, albedo=False)
    print(data.keys(), data["rgb"].shape, data["depth"].shape, data["segmentation"].shape)
    env.close()

```

YZY14606, 2025-03-18:
Okay! Thanks very much!