mani-skill/ManiSkill #922: vk::PhysicalDevice::createDeviceUnique: ErrorInitializationFailed Error Despite vkcube Rendering

sagnikanupam, 2025-03-10:
Hi there,

We're trying to setup the ManiSkill docker image on our system for headless rendering, and we keep running into a `RuntimeError: vk::PhysicalDevice::createDeviceUnique: ErrorInitializationFailed` error whenever the code calls `sapien.render.RenderSystem()`. We would really appreciate any help you could provide us regarding the installation.

## System Description:

Debian GNU/Linux 11 (bullseye)
8 NVIDIA RTX A6000 GPUs
CUDA 12.6
Driver Version: 560.35.05

## Errors

```
docker pull maniskill/base
docker run --rm -it --gpus all maniskill/base python -m mani_skill.examples.demo_random_action
````

results in

```
 File "/opt/conda/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/opt/conda/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/examples/demo_random_action.py", line 128, in <module>
    main(parsed_args)
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/examples/demo_random_action.py", line 83, in main
    env: BaseEnv = gym.make(
  File "/opt/conda/lib/python3.9/site-packages/gymnasium/envs/registration.py", line 802, in make
    env = env_creator(**env_spec_kwargs)
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/utils/registration.py", line 182, in make
    env = env_spec.make(**kwargs)
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/utils/registration.py", line 79, in make
    return self.cls(**_kwargs)
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/envs/tasks/tabletop/push_cube.py", line 65, in __init__
    super().__init__(*args, robot_uids=robot_uids, **kwargs)
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/envs/sapien_env.py", line 338, in __init__
    obs, _ = self.reset(seed=[2022 + i for i in range(self.num_envs)], options=dict(reconfigure=True))
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/envs/sapien_env.py", line 835, in reset
    self._reconfigure(options)
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/envs/sapien_env.py", line 671, in _reconfigure
    self._setup_scene()
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/envs/sapien_env.py", line 1108, in _setup_scene
    sapien.Scene([physx_system, sapien.render.RenderSystem(self._render_device)])
RuntimeError: vk::PhysicalDevice::createDeviceUnique: ErrorInitializationFailed
```

Additionally, 

```
docker run --rm -it --gpus all maniskill/base python -m sapien.example.offscreen
```
also results in a similar error.

```
Traceback (most recent call last):
  File "/opt/conda/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/opt/conda/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/opt/conda/lib/python3.9/site-packages/sapien/example/offscreen.py", line 37, in <module>
    main()
  File "/opt/conda/lib/python3.9/site-packages/sapien/example/offscreen.py", line 9, in main
    scene = sapien.Scene()
  File "/opt/conda/lib/python3.9/site-packages/sapien/wrapper/scene.py", line 25, in __init__
    [sapien.physx.PhysxCpuSystem(), sapien.render.RenderSystem()]
RuntimeError: vk::PhysicalDevice::createDeviceUnique: ErrorInitializationFailed
```

## Vulkan

Vulkan seems to be working; `ssh -X`ing (with XQuartz) onto the machine and running `vkcube` renders a cube, and we have our vulkanutils output posted here: [vulkaninfo.txt](https://github.com/user-attachments/files/19170521/vulkaninfo.txt)

## ManiSkill Installation Troubleshooting Files
For the files mentioned in ManiSkill installation, on the host machine we have:
/usr/share/vulkan/icd.d/nvidia_icd.json
```
{
    "file_format_version" : "1.0.0",
    "ICD" : {
        "library_path" : "libEGL_nvidia.so.0"
    }
}
```
/usr/share/glvnd/egl_vendor.d/10_nvidia.json
```
{
    "file_format_version" : "1.0.1",
    "ICD": {
        "library_path": "libGLX_nvidia.so.0",
        "api_version" : "1.3.280"
    }
}
```
/etc/vulkan/implicit_layer.d/nvidia_layers.json
```
{
    "file_format_version" : "1.0.0",
    "layer": {
        "name": "VK_LAYER_NV_optimus",
        "type": "INSTANCE",
        "library_path": "libGLX_nvidia.so.0",
        "api_version" : "1.2.155",
        "implementation_version" : "1",
        "description" : "NVIDIA Optimus layer",
        "functions": {
            "vkGetInstanceProcAddr": "vk_optimusGetInstanceProcAddr",
            "vkGetDeviceProcAddr": "vk_optimusGetDeviceProcAddr"
        },
        "enable_environment": {
            "__NV_PRIME_RENDER_OFFLOAD": "1"
        },
        "disable_environment": {
            "DISABLE_LAYER_NV_OPTIMUS_1": ""
        }
    }
}
```

StoneT2000, 2025-03-11:
@fbxiang any idea?

fbxiang, 2025-03-11:
The content of  /usr/share/vulkan/icd.d/nvidia_icd.json and /usr/share/glvnd/egl_vendor.d/10_nvidia.json seem swapped (probably a pasting error?). Otherwise I do not really see a problem.
Maybe you could try to run SAPIEN's test outside of Docker and see if `nvidia_icd.json` is properly mapped into the container.

StoneT2000, 2025-03-14:
The first potential fix is to reinstall nvidia drivers completely on the machine by install nvidia-driver-xxx package via apt install or something. (do not use run file and don't install nvidia-driver-server related ones). `nvidia-driver-535` works for my ubuntu machine.

sabari245, 2025-09-09:
I'm in a position where I can't use GPU for simulation. I must run this entirely on CPU. Has anyone successfully run ManiSkill (with the SAPIEN backend) using only CPU? Any pointers appreciated.