mani-skill/ManiSkill #925: Issue: Hand Simulation in ManiSkill Leads to NaN Positions

lifchrs, 2025-03-12:
I'm trying to simulate a hand from Inspire Robotics, and as a basic sanity check, I'm sampling random actions to observe its behavior. However, after a few seconds, the fingers fly to NaN positions.

- I'm using the [Dex-Suite URDF](https://github.com/dexsuite/dex-urdf/blob/main/robots/hands/inspire_hand/inspire_hand_right.urdf)
- The active joints are controlled via a `PDJointPosController`.
- The mimic joints are controlled using a passive controller.

Observations:
- This issue persists even when the collision meshes are commented out in the URDF.
- The same URDF runs on MuJoCo without running into this issue, which suggests this could be a bug in ManiSkill.

You can see a video of this issue happening [here](https://drive.google.com/file/d/1zF_KPb7GX01p4Rs-c_qTP0lZJ-Ghk4Nm/view?usp=sharing).

I've also created a minimal GitHub repository containing that reproduces this behavior:
**[https://github.com/lifchrs/dexterous-hand](https://github.com/lifchrs/dexterous-hand)**

Would appreciate any insights or suggestions on how to resolve this. Thanks!

StoneT2000, 2025-03-13:
@lifchrs can you share your agent class you used for this?

lifchrs, 2025-03-13:
It's [here](https://github.com/lifchrs/dexterous-hand/blob/915ab575d650925e015cf302c6087a45a1e0ce01/hand.py#L11) in the repository I linked.

lifchrs, 2025-03-20:
You're correct - I just pushed a fix to that. Unfortunately, fixing that doesn't solve the issue at all though. Out of curiosity, are you also trying to use an Inspire hand? And if so, are you using the same urdf as I am?

Updated video:

https://github.com/user-attachments/assets/909645a3-1366-4fab-81b1-531bb24213ca

lifchrs, 2025-03-20:
What script are you running?

StoneT2000, 2025-03-20:
I have confirmed something doesn't seem to work right. our lab has our own inspire hand setup in maniskill that we used for our teleop work that is being open sourced. Let me see what the differences are.

StoneT2000, 2025-03-20:
Ah you might be right @m-vik 

I think the support for mimic tags is there but it was implemented maybe poorly.

StoneT2000, 2025-03-20:
Ah okay it seems the one we tried before we also disabled all the mimic joints and instead used a mimic joint controller which essentially sends the same joint control signal to all given joints. I can share an example config of how to do so.

This kind of issue is not documented so will need to update our docs for that.

I will also discuss with the SAPIEN people to see how physx is meant to correctly handle mimic joints. I believe we encounter similar issue before

lifchrs, 2025-03-20:
Thanks for the update! An open-sourced inspire hand would be great! Do you have an approximate timeline for that? 

To make sure I'm understanding you correctly, by disabling mimic joints you mean that you just commented out lines involving mimic joints in the URDF? So the issue is with how mimic joints are loaded from the URDF, but using a mimic joint position controller works, right?

StoneT2000, 2025-03-20:
Yes although I notice the one in ManiSkill for mimic joint position control is a bit simple. Currently updating it so you can pass in properties like multiplier and offset.

I will upload a floating inspire hand + one attached to xarm

lifchrs, 2025-03-24:
@StoneT2000 would it be possible for you to still share your current inspire hand setup and controller config?

StoneT2000, 2025-03-27:
Sorry still working on it. It requires modifyign some of the existign code in maniskill so still testing it. I am also testing it a little bit on the real hand have to make sure things make sense.

StoneT2000, 2025-04-04:
@lifchrs Ok with the latest git commit you can experiment with inspire hand + mimic controller setup now (which is stable).

Next pypi release for maniskill nightly will be tomorrow, you can wait for that if you wish for a pip install version.

Example code:

```
python mani_skill/examples/demo_robot.py -r floating_inspire_hand_right --random-actions -c pd_joint_delta_pos
```

https://github.com/haosulab/ManiSkill/blob/b0485f0f9616a25c26b3843c24f11185edfe5166/mani_skill/agents/robots/floating_inspire_hand/floating_inspire_hand.py implementation here

StoneT2000, 2025-04-22:
1 degree of error sounds kind of bad, 0.1 is more acceptable.

I'll ask the people in my lab who tested inspire hand and tried sim2real and see what they had encountered.

In my opinion however while 1 degree is somewhat big, it might not even affect your work

(you probably should not feed the joint angles of passive joints to your policy anyway).

StoneT2000, 2025-05-20:
I may try sometime next week to see if there are any other modelling options for this hand to be a bit more accurate. We had some internal testing recently with physx to try some other options.

StoneT2000, 2025-05-20:
Oh and if you run

```
 python mani_skill/examples/demo_robot.py -r floating_inspire_hand_right --none-actions -c pd_joint_pos -b physx_cpu
```

Click the robot and find one of the control joints and open the collapsed section to change the position target. The simulator should try and follow that with the mimic joints and you can see live in the GUI the mimic joint in action.

Let me know how it goes and whether the error is acceptable.

StoneT2000, 2025-05-20:
Ok just updated the PR and merged it, it should be entirely fixed now. Needed unfortunately a few small hacks but simulation on CPU and GPU looks stable! And almost no error. It makes a few possibly breaking changes to the controller code so you will need to check things again probably. Fixed based version and left hand version are also added now.

closed 2025-05-20 by StoneT2000