mani-skill/ManiSkill #935: [Bug] Renderer window crashing when clicking

alexcbb, 2025-03-14:
Hello, 

I've been using recently the Viewer to explore some scenes with the "human_render" to explore a bit but it seems that the window always crash when I try to click on it.

Here is the error :   
```
File "~/miniconda3/envs/cond_env/lib/python3.11/site-packages/sapien/utils/viewer/control_window.py", line 389, in _handle_click
    tw, th = self.window.get_picture_size("Segmentation")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
IndexError: _Map_base::at
```
It  seems to come from SAPIEN, I don't know whether I'm on the good repo. 

I'm in Ubuntu 22.04
The version I'm using are sapien 3.0.0b1, python 3.11.11 and mani_skill 3.0.0b19.

Thank you in advance for your help.

alexcbb, 2025-03-17:
I'm using a custom script based on the Maniskill-HAB env : https://arth-shukla.github.io/mshab/

StoneT2000, 2025-03-17:
Can you share the script with gym.make and its arguments?

closed 2025-04-27 by StoneT2000

jstmn, 2025-05-19:
I'm also experiencing this. I've determined that this behavior happens only when there's been a second environment created. In this script for example, once you add `env2`, you get the error below. It doesn't seem to matter what device the second environment is created with btw - same error with cpu vs. gpu

```
import gymnasium as gym
import mani_skill.envs
import numpy as np

env = gym.make(
    "PickCube-v1",
    num_envs=10,
    obs_mode="state",
    control_mode="pd_ee_delta_pose",
    render_mode="human",
    sim_backend="gpu",
    parallel_in_single_scene=True,
)
env2 = gym.make(
    "PickCube-v1",
    obs_mode="state",
    control_mode="pd_ee_delta_pose",
    render_mode="human",
    num_envs=1,
    sim_backend="cpu",
    # num_envs=10,
    # sim_backend="gpu",
    # parallel_in_single_scene=True,
)

obs, _ = env.reset(seed=0) # reset with a seed for determinism
done = False
while not done:
    obs, reward, terminated, truncated, info = env.step(env.action_space.sample())
    env.render()  # a display is required to render
env.close()
```

```
[2025-05-19 16:14:03.271] [SAPIEN] [warning] A PhysX CPU system is being created while PhysX GPU is enabled. You can safely ignore this message if it is intended. To use GPU PhysX, create a sapien.physx.PhysxGpuSystem explicitly and pass it to sapien.Scene constructor.
Traceback (most recent call last):
  File "/home/jstm/Projects/mpcm2/scripts/click_failing.py", line 34, in <module>
    env.render()  # a display is required to render
  File "/home/jstm/Projects/mpcm2/.venv/lib/python3.10/site-packages/gymnasium/core.py", line 337, in render
    return self.env.render()
  File "/home/jstm/Projects/mpcm2/.venv/lib/python3.10/site-packages/gymnasium/wrappers/common.py", line 409, in render
    return super().render()
  File "/home/jstm/Projects/mpcm2/.venv/lib/python3.10/site-packages/gymnasium/core.py", line 337, in render
    return self.env.render()
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/envs/sapien_env.py", line 1373, in render
    return self.render_human()
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/envs/sapien_env.py", line 1308, in render_human
    self._viewer.render()
  File "/home/jstm/Projects/mpcm2/.venv/lib/python3.10/site-packages/sapien/utils/viewer/viewer.py", line 228, in render
    plugin.after_render()
  File "/home/jstm/Projects/mpcm2/.venv/lib/python3.10/site-packages/sapien/utils/viewer/control_window.py", line 364, in after_render
    self._handle_click()
  File "/home/jstm/Projects/mpcm2/.venv/lib/python3.10/site-packages/sapien/utils/viewer/control_window.py", line 388, in _handle_click
    tw, th = self.window.get_picture_size("Segmentation")
IndexError: _Map_base::at
```

@StoneT2000 any idea how to fix this?

StoneT2000, 2025-05-20:
What did you click exactly? Is it anywhere? Or one of the objects?

jstmn, 2025-05-20:
[image]

Seems to be anywhere in the scene. The settings panels are safe.