mani-skill/ManiSkill #938: Issue with Using Custom Camera Intrinsics

SumeetBatra, 2025-03-18:
Hi all,

I am having some issues with the images Maniskill returns when I set the camera intrinsics parameter. I have the intrinsics matrix for Intel Realsense D515 for a 640x480 image following the standard convention [fx 0 cx 0 fy cy 0 0 1]. These are the parameters just fyi:

`D515
Width: 640
Height: 480
Focal Length: (601.3878173828125, 601.7159423828125)
Principal Point: (333.2942199707031, 245.56826782226562)`

I want to receive 128x128 image observations from the simulator so I rescale the focal lengths and principle points like so: 

```
scale_x, scale_y = 0.2, 0.26667  # for rescaling to 128x128
intrinsics = [601.3878173828125 * scale_x, 0, 333.2942199707031 * scale_x, 0., 601.7159423828125 * scale_y, 245.56826782226562 * scale_y, 0., 0., 1.]
```

and then I pass the following sensor config dict to maniskill:

```
    sensor_configs = {
        'base_camera': {
            'intrinsic': np.array([intrinsics] * num_envs).reshape(num_envs, 3, 3),
            'width': 128,
            'height': 128,
            'fov': None,
            'pose': look_at(eye=[0.3, 0, 0.6], target=[-0.1, 0, 0.1])  # default
        }  
    }
```
However, here's what a rollout with these parameters looks like: 

https://github.com/user-attachments/assets/06fb4c40-144c-45c3-9b93-14d1c2b02b50

And for reference, this is what happens when I don't set the intrinsics matrix and use the same look_at values:

https://github.com/user-attachments/assets/ad832d7c-b9b7-4c2b-a5af-51c866af4699

Now I'm not an expert in cameras, but to me these look completely different-- beyond what different intrinsic parameters would cause. Am I doing something wrong with this pipeline, or is this intended behavior? Thanks!

StoneT2000, 2025-03-18:
intel realsense cameras have pretty small fovs and look a lot more zoomed in. I am not surprised it is this close. It might look a more squished width wise since you do not do a center crop with the intrinsics (but if you also resize your real camera the same way you did with the sim one to 128x128 it should look similar).

If you don't want the width wise squishing. You can first do a center crop then scale to 128x128. This means you should subtract 640-480=240 (size of the longer side difference with smaller side which you crop out) from the x-axis principal point). Then scale down by 480/128 on all the values in your intrinsic matrix (except the 1. value). 

https://stackoverflow.com/questions/74749690/how-will-the-camera-intrinsics-change-if-an-image-is-cropped-resized explains a bit as well.

SumeetBatra, 2025-03-18:
Thanks for the reply, although I'm still a bit confused on a few points:

1. I don't do resizing in Maniskill, I just pass the height and width parameters directly as 128 in the sensor config. Is Maniskill internally doing resizing from some other natively rendered resolution, or will it natively render at 128x128?
2. When you say center crop and rescale the image and the intrinsics, do you mean pass in 640x480 as the H/W params to maniskill, and then manually crop/rescale myself, as opposed to natively rendering at 128x128 (if that's what's actually happening)?

SumeetBatra, 2025-03-18:
Ok that makes sense. Thanks!

closed 2025-03-18 by SumeetBatra