mani-skill/ManiSkill #939: Request for Flow Matching Baseline

yeknafar, 2025-03-18:
First of all, I sincerely appreciate your incredible work on ManiSkill! Your efforts in providing high-quality benchmarks and simulation environments for robotic manipulation are truly valuable to the community.

I would like to request the addition of a Flow Matching (FM) baseline in ManiSkill. Given that the [π₀ model](https://www.[physicalintelligence.company/download/pi0.pdf](https://www.physicalintelligence.company/download/pi0.pdf)) uses Flow Matching, having an official FM baseline would be highly beneficial. It would provide a strong foundation for comparison and further research in imitation learning and reinforcement learning.

Could you consider adding this baseline in future updates?

Best regards,

StoneT2000, 2025-03-18:
We would love to have such a baseline but this is too much work for our own team. We are happy to accept some contributions along this line that takes pi0 style model / FM training approach and tests it on simulated datasets and environments.

I think a starting point to make this work is if someone wants to find a way to convert maniskill demo datasets to whatever format the pi0 codebase uses and finetunes pi0 on those datasets and then evaluates it in simulation (with a pd_joint_target_pos or pd_joint_pos controller probably)

yeknafar, 2025-03-18:
Thank you for your response and for sharing your thoughts on this! I truly appreciate the effort your team has put into ManiSkill and understand that adding a Flow Matching baseline requires significant work.

I will be working on Flow Matching in the next few months, and if I achieve any promising results, I would be happy to share them with the community. I look forward to exploring this direction and contributing if possible.

Thanks again for your time and insights!

StoneT2000, 2025-03-18:
Sounds great, please keep us posted here on progress. If the baseline code is clean and works okay we can add it to the repo's examples/baselines folder.

As a recommendation you can start with the basic PushCube and then PickCube tasks

awesomeNabi, 2025-03-28:
Really looking forward FM baselines to be added in Mainskill!

Mikasa0503, 2026-09-21:
Hi @StoneT2000, is this still available? I'd like to work on a scoped first version.

My plan is to add a state-observation flow-matching baseline under `examples/baselines`, starting with `PushCube-v1` and the existing demonstration data. It would use action chunks and Euler integration at inference, and include the training/evaluation config, focused tests for the flow target and integration path, and results from three seeds. If it works reliably on PushCube, I'll also run it on PickCube and report those results.

For the first PR I'd leave pi0/VLM integration and image observations out. The goal would be a small, readable reference implementation that can be compared with the existing BC and diffusion-policy baselines.

I checked the issue timeline and open PRs and didn't find another FM implementation in progress. I also saw that @yeknafar planned to explore this earlier, so I'm happy to coordinate if that work is still ongoing. Does this scope sound useful?