mani-skill/ManiSkill #940: Upgrade gymnasium to 1.X

JesseFarebro, 2025-03-18:
Given that ManiSkill3 is still in prerelease maybe it's a good time to upgrade to Gymnasium 1.X. Many of the other environments we're employing are using 1.X and ManiSkill is our only conflict.

StoneT2000, 2025-03-18:
I'll keep this issue up as the main feature request. This is now the 2nd/3rd time someone has requested for an upgrade to gymnasium so I'm now giving this a more stronger consideration than before

The original discussion was here: https://github.com/haosulab/ManiSkill/discussions/933 but we can keep it alive in this issue instead

copying my original response

> Definitely not supporting gymnasium v1.0 due to the breaking API change for vector env interfaces.
However I just dug through their blog posts and noticed they have a more compatible v1.1 which now enables support of some kind same step resetting (the current RL standard and approach to auto reset envs)
Gym >= 1.0 also has some nice improvements like clear separation of vector vs not vector envs as well as vector vs non vector wrappers which would now be usable with GPU sims (single env wrappers do not work unless you use cpu sim + multiprocessing based parallelization)
I think long term we will eventually upgrade but I suspect it might take some time because some code in maniskill leverages the gymnasium structure to be as interoperable with most RL code as possible. Moreover I think most RL code bases do not support using later versions of gymnasium yet and so I don't see this as being extremely urgent.
The other thing is the improvements made by gym 1.1 do not have any impact on RL in ManiSkill or environment speeds in general afaik so there's also no reason there to upgrade yet.

The main blockers for us right now is mostly time constraints probably. The other concern is trying to see how easy it is to ensure version 0.29 still works for users while supporting >=1.1 gymnasium. 

@JesseFarebro do you know what environments have already performed this migration that are not maintained by Farama foundation? And more importantly what RL libraries have done this migration already? I wish to know primarily because I want to see how these other libraries are planning to maintain backwards compatability.

StoneT2000, 2025-05-28:
Ok it looks like the RL baselines are running okay now. Try out the code on this branch: gym-1.1-support