mani-skill/ManiSkill #946: Get the 7DoF action

Jimntu, 2025-03-19:
May I ask how can I get the 7DoF action output (x,y,z,r,p,y,gripper) for each step when the robotic arm is performing the task? The "action" stored in h5 file is a 8-dim vector, which is not the 7DoF information needed in general VLA. A convenient way to fetch the 7DoF information is necessary.

Jimntu, 2025-03-19:
I believe I need to change the controller from "pd_joint_pos" to "pd_ee_delta_pose" to get the end-effector 7DoF actions. However, when I change the controller, this error occured.

assert action.shape == (
AssertionError: Received action of shape torch.Size([8]) but expected shape (1, 7)

How to modify other codes when I would like to change the controller mode? Thanks in advance!

StoneT2000, 2025-03-19:
How are you changing the controller?

did you do `gym.make(..., control_mode="pd_ee_delta_pose")` that action space should be size (n, 7) for n environments since each action is x,y,z,r,p,y,gripper.

Jimntu, 2025-03-20:
Really appreciate your prompt response!

Yes, I did the `gym.make(..., control_mode="pd_ee_delta_pose") `and action.shape is still 8. I think the problem is in motionplanner.py  L95  `action = np.hstack([qpos, self.gripper_state])`, this returns the 8-dim variable (qpos 7-dim + gripper 1-dim) rather than x,y,z,r,p,y,gripper as long as the control mode is not` "pd_joint_pos_vel"`

```
if self.control_mode == "pd_joint_pos_vel":
                qvel = result["velocity"][min(i, n_step - 1)]
                action = np.hstack([qpos, qvel, self.gripper_state])
else:
                action = np.hstack([qpos, self.gripper_state]) #7+1

```
Do I miss something?

StoneT2000, 2025-03-20:
You are using motion planning? Motion planners generate joint position/velocity controls, not end effector controls. This is thus not really compatible. You have to generate actions in the same action space.

Jimntu, 2025-03-20:
Yes, I am using motionplanner to perform the pre-defined tasks in mani_skill/examples/motionplanning/panda/solutions/pick_cube.py

Could you suggest another way to perform these manipulation tasks other than motionplanning so that I could get 7 Dof action for every step?

Thank you so much!

Jimntu, 2025-03-21:
Let me rephrase the question: Does ManiSkill provide example demonstrations or solutions using the pd_ee_delta_pose controller, similar to how it does for the pick_cube task in motion planning?

Jimntu, 2025-03-21:
Yes, I am using the motion planner.

Since I understand that the motion planner does not directly output 7-DoF actions, I’d like to ask:
Does ManiSkill provide any example demonstrations or solutions using the pd_ee_delta_pose controller—similar to how it provides demonstrations for the pick_cube task under the motion planning examples?

Alternatively, is there a way to **retrieve the 7-DoF end-effector pose (or actions) at each step** while using the motion planner? I’m aiming to collect image-action pairs for each step to train a VLA (Visual Language Action) model.

Any suggestions or guidance would be incredibly helpful. Thank you so much!

Jimntu, 2025-03-22:
I find out a trick that could retrieve 7 DoF action for every step while using motion planing

For each time env.step(action):

```
pose = self.env.agent.tcp.pose
xyz = pose.p[0]
quat = pose.q[0]
rpy = R.from_quat([quat[1], quat[2], quat[3], quat[0]]).as_euler("xyz")  # EE orientation: convert quaternion [w, x, y, z] -> [x, y, z, w]
ee_pose = np.concatenate([xyz, rpy, [self.gripper_state]])
print("7-DoF EE Action:", ee_pose)
```

StoneT2000, 2025-03-25:
This is not the same as the action to take. It might sometimes work but the current TCP pose is not necessarily the action that will achieve that pose in the next timestep.

We usually use a replay script like this to verify whether our action conversion works or not: https://maniskill.readthedocs.io/en/latest/user_guide/datasets/replay.html#replaying-converting-trajectories

it does a number of other things as well such as checking the error after conversion. We recommend using the existing replay trajectory tool

Jimntu, 2025-03-25:
If I use the TCP pose of next timestep minus the TCP pose of this timestep, can I get the delta EE action this time?

Lr-2002, 2025-04-17:
Is there some other we to get the ideal  traj from the joint_pos to delta pose ?

StoneT2000, 2025-04-17:
See how we do it. It's a little complicated: https://github.com/haosulab/ManiSkill/blob/147e0677f264b39f417335402861c1847b066563/mani_skill/trajectory/replay_trajectory.py#L328-L338

and

https://github.com/haosulab/ManiSkill/blob/main/mani_skill/trajectory/utils/actions/conversion.py#L217

Lr-2002, 2025-04-18:
wow, Thanks for your replay. 
I tried replay, it works well !

closed 2025-04-18 by StoneT2000