mani-skill/ManiSkill #950: How to Create My Own Scene

jzx-gooner, 2025-03-20:
First of all, you've done an amazing job, and thank you for sharing. I exported a URDF, which is essentially a cabinet with a knob on it. I want to create a scene that includes this cabinet and a panda_wrist_camera robot. Which script should I refer to in order to achieve this? Thank you!

StoneT2000, 2025-03-25:
For accessing link/joint data see below
```
articulation = builders.build(name="my_articulation")
print(articulation.joints_map.keys())
articulation.joints_map["knob_joint"] # returns a Joint object, you can then access e.g.  the joint position via .qpos
articulation.links_map["knob"] # returns Link object, you can then access e.g. .pose
```

Is this GPU sim with 16 parallel envs? Loading very complex objects can slow down simulation, what is in this URDF that you build? What other objects are there? Are you rendering as well?

StoneT2000, 2025-05-13:
Is this still an issue @jzx-gooner ?

closed 2025-05-20 by StoneT2000