mani-skill/ManiSkill #952: SAPIEN tab not showing anything

ThisisXXZ, 2025-03-20:
Platform: Windows 11
I've installed ManiSkill using the following commands: 
```
conda create -n ManiSkill python=3.9
conda activate ManiSkill
pip install mani_skill-nightly torch
conda install pinocchio -c conda-forge
```

Then I ran the command
```
python -m mani_skill.examples.demo_random_action -e PickCube-v1 --render-mode="human" --shader="rt-fast"
```
to play with the demo.
A SAPIEN tab pops up, but it's all white and not showing anything. 
The output is: 
```
(ManiSkill) PS C:\Users\xxz03\Desktop\Document\files\playground> python -m mani_skill.examples.demo_random_action -e PickCube-v1 --render-mode="human" --shader="rt-fast"     
Unable to read VR Path Registry from C:\Users\xxz03\AppData\Local\openvr\openvrpaths.vrpath
Unable to read VR Path Registry from C:\Users\xxz03\AppData\Local\openvr\openvrpaths.vrpath
Observation space Dict()
Action space Box(-1.0, 1.0, (8,), float32)
Control mode pd_joint_delta_pos
Reward mode normalized_dense
```
Am I missing something here? Thank you so much!

StoneT2000, 2025-03-20:
Is this via WSL or using windows? Looks like just windows

Can you first try running 
```
pip show sapien
```

and
```
python -m sapien.example.hello_world
```

If that doesn't work please raise an issue on the SAPIEN github https://github.com/haosulab/SAPIEN

ThisisXXZ, 2025-03-20:
> Is this via WSL or using windows? Looks like just windows
> 
> Can you first try running
> 
> ```
> pip show sapien
> ```
> 
> and
> 
> ```
> python -m sapien.example.hello_world
> ```
> 
> If that doesn't work please raise an issue on the SAPIEN github https://github.com/haosulab/SAPIEN

It's pure Windows, yes.
The result of pip show: 
```
Name: sapien
Version: 3.0.0b1
Summary: ['SAPIEN: A SimulAted Parted based Interactive ENvironment']
Home-page: https://sapien.ucsd.edu
Author: Sapien
Author-email: sapienaicontact@gmail.com
License: MIT
Location: c:\users\xxz03\anaconda3\envs\maniskill\lib\site-packages
Requires: lxml, networkx, numpy, opencv-python, pyperclip, requests, transforms3d
Required-by: mani-skill-nightly
```

And the SAPIEN can run correctly with ``sapien.example.hello_world``

[image]

StoneT2000, 2025-03-20:
Can you try running

```
import mani_skill.envs
import gymnasium as gym

env = gym.make("PickCube-v1", render_mode="human", obs_mode="state")
obs, _ = env.reset(seed=0, options=dict(reconfigure=True))
for _ in range(50):
  env.step(env.action_space.sample())
  env.render()
```

As well as

```
import mani_skill.envs
import gymnasium as gym

env = gym.make("PickCube-v1", obs_mode="state")
obs, _ = env.reset(seed=0, options=dict(reconfigure=True))
for _ in range(50):
  env.step(env.action_space.sample())
  env.unwrapped.render_human()
```

StoneT2000, 2025-03-20:
Oh it might be because of the shader used. Try original script with the default shader?

And the tab closes because the script exits

ThisisXXZ, 2025-03-21:
> Oh it might be because of the shader used. Try original script with the default shader?
> 
> And the tab closes because the script exits

Thanks! The default shader works well.

closed 2025-03-21 by ThisisXXZ