mani-skill/ManiSkill #953: Using StereoDepthCameraConfig produces AttributeError

BlGene, 2025-03-20:
In my code, I am trying to substitute StereoDepthCameraConfig with CameraConfig (my env inherits from `BaseEnv`)

```
        #self.render_camera_config = CameraConfig("render_camera", pose,  self.cam_size,  self.cam_size, 1, 0.01, 100)
        self.render_camera_config = StereoDepthCameraConfig("render_camera", pose,  self.cam_size,  self.cam_size, 1, 0.01, 100)

    @property
    def _default_human_render_camera_configs(self):
        return self.render_camera_config
```
This gives the error:

```
  File "/home/X/ManiSkill/mani_skill/envs/sapien_env.py", line 828, in reset
    self._reconfigure(options)
  File "/home/X/ManiSkill/mani_skill/envs/sapien_env.py", line 671, in _reconfigure
    self._setup_sensors(options)
  File "/home/X/ManiSkill/mani_skill/envs/sapien_env.py", line 747, in _setup_sensors
    self._sensors[uid] = sensor_cls(
  File "/home/X/ManiSkill/mani_skill/sensors/depth_camera.py", line 71, in __init__
    self.camera = StereoDepthSensor(
  File "/home/X/miniconda3/envs/maniskill/lib/python3.10/site-packages/sapien/sensor/stereodepth.py", line 146, in __init__
    self._create_cameras()
  File "/home/X/local/miniconda3/envs/maniskill/lib/python3.10/site-packages/sapien/sensor/stereodepth.py", line 383, in _create_cameras
    self._cam_rgb = RenderCameraComponent(*self._config.rgb_resolution)
AttributeError: 'str' object has no attribute 'rgb_resolution'
```

I can re-produce it with the following code:
```
import sapien.core as sapien
from sapien.sensor import StereoDepthSensor, StereoDepthSensorConfig

sim = sapien.Engine()
renderer = sapien.SapienRenderer()
sim.set_renderer(renderer)
scene_config = sapien.SceneConfig()
scene = sim.create_scene(scene_config)
sensor_config = StereoDepthSensorConfig()
sensor = StereoDepthSensor('sensor', scene, sensor_config)
```

>>> sapien.__version__
'3.0.0b1'
>>> import mani_skill
mani_skill.__version__
>>> mani_skill.__version__
'3.0.0b20'
>>> 

Do you have any tips?

Note, I made some slight changes:
```
--- a/mani_skill/sensors/depth_camera.py
+++ b/mani_skill/sensors/depth_camera.py
@@ -40,7 +40,7 @@ class StereoDepthCamera(Camera):
         self,
         camera_cfg: StereoDepthCameraConfig,
         scene: sapien.Scene,
-        renderer_type: str,
+        renderer_type: str = "sapien",
         articulation: physx.PhysxArticulation = None,
     ):
         self.camera_cfg = camera_cfg
@@ -69,7 +69,7 @@ class StereoDepthCamera(Camera):
         sensor_config.min_depth = camera_cfg.min_depth
         if self.actor is None:
             self.camera = StereoDepthSensor(
-                camera_cfg.uid, scene, sensor_config, mount=self.actor
+                camera_cfg.uid, scene, sensor_config, #mount=self.actor
             )
             self.camera.set_pose(camera_cfg.pose)
         else:
```

StoneT2000, 2025-04-08:
The StereoDepthCameraConfig is currently not supported just yet, the setup for it does not work just yet unfortunately. We can add it back in the future.

akushonkamen, 2026-10-20:
I'd like to work on this. Plan: implement the missing sensor registration path for `StereoDepthCameraConfig` so it at least parses/configures correctly on the CPU sim backend (stereo RGB + depth outputs), starting from the existing `CameraConfig` implementation.