mani-skill/ManiSkill #955: 'pd_ee_pose', 'pd_ee_delta_pose', and 'pd_ee_target_delta_pose' are super slow for panda

jstmn, 2025-03-21:
Hi,

I've noticed that the 'PD_EE_POSE', 'PD_EE_TARGET_DELTA_POS', and 'PD_EE_TARGET_DELTA_POSE' controllers are quite slow for Panda. Taking random actions with the 'PD_EE_POSE' controller takes ~5x as long as a joint space controller. Is there someway to speed them up? 

I looked through the controller code but didn't see anything obvious. I imagine the IK calculation is slow

Here's my timing data. This is running with a 4090 with gpu mode enabled.
```
Evaluating 'PD_JOINT_DELTA_POS'
16 envs	  -> t_elapsed: 1.699361 seconds | t_elapsed_per_step: 0.016994 s/step
32 envs	  -> t_elapsed: 2.464377 seconds | t_elapsed_per_step: 0.024644 s/step
64 envs	  -> t_elapsed: 3.02493 seconds | t_elapsed_per_step: 0.030249 s/step

Evaluating 'PD_JOINT_POS'
16 envs	  -> t_elapsed: 1.687097 seconds | t_elapsed_per_step: 0.016871 s/step
32 envs	  -> t_elapsed: 2.150127 seconds | t_elapsed_per_step: 0.021501 s/step
64 envs	  -> t_elapsed: 2.631921 seconds | t_elapsed_per_step: 0.026319 s/step

Evaluating 'PD_EE_DELTA_POS'
16 envs	  -> t_elapsed: 2.618853 seconds | t_elapsed_per_step: 0.026189 s/step
32 envs	  -> t_elapsed: 3.061343 seconds | t_elapsed_per_step: 0.030613 s/step
64 envs	  -> t_elapsed: 3.587767 seconds | t_elapsed_per_step: 0.035878 s/step

Evaluating 'PD_EE_DELTA_POSE'
16 envs	  -> t_elapsed: 2.917914 seconds | t_elapsed_per_step: 0.029179 s/step
32 envs	  -> t_elapsed: 3.339032 seconds | t_elapsed_per_step: 0.03339 s/step
64 envs	  -> t_elapsed: 3.669804 seconds | t_elapsed_per_step: 0.036698 s/step

Evaluating 'PD_EE_POSE'
16 envs	  -> t_elapsed: 23.181712 seconds | t_elapsed_per_step: 0.231817 s/step
32 envs	  -> t_elapsed: 26.639849 seconds | t_elapsed_per_step: 0.266398 s/step
64 envs	  -> t_elapsed: 25.808958 seconds | t_elapsed_per_step: 0.25809 s/step

Evaluating 'PD_JOINT_TARGET_DELTA_POS'
16 envs	  -> t_elapsed: 2.132394 seconds | t_elapsed_per_step: 0.021324 s/step
32 envs	  -> t_elapsed: 2.634803 seconds | t_elapsed_per_step: 0.026348 s/step
64 envs	  -> t_elapsed: 2.999529 seconds | t_elapsed_per_step: 0.029995 s/step

Evaluating 'PD_EE_TARGET_DELTA_POS'
16 envs	  -> t_elapsed: 14.729411 seconds | t_elapsed_per_step: 0.147294 s/step
32 envs	  -> t_elapsed: 15.948367 seconds | t_elapsed_per_step: 0.159484 s/step
64 envs	  -> t_elapsed: 19.013234 seconds | t_elapsed_per_step: 0.190132 s/step

Evaluating 'PD_EE_TARGET_DELTA_POSE'
16 envs	  -> t_elapsed: 14.266906 seconds | t_elapsed_per_step: 0.142669 s/step
32 envs	  -> t_elapsed: 16.796258 seconds | t_elapsed_per_step: 0.167963 s/step
64 envs	  -> t_elapsed: 19.147189 seconds | t_elapsed_per_step: 0.191472 s/step

Evaluating 'PD_JOINT_VEL'
16 envs	  -> t_elapsed: 2.446301 seconds | t_elapsed_per_step: 0.024463 s/step
32 envs	  -> t_elapsed: 2.535609 seconds | t_elapsed_per_step: 0.025356 s/step
64 envs	  -> t_elapsed: 3.054398 seconds | t_elapsed_per_step: 0.030544 s/step

Evaluating 'PD_JOINT_POS_VEL'
16 envs	  -> t_elapsed: 2.318194 seconds | t_elapsed_per_step: 0.023182 s/step
32 envs	  -> t_elapsed: 1.912065 seconds | t_elapsed_per_step: 0.019121 s/step
64 envs	  -> t_elapsed: 2.436522 seconds | t_elapsed_per_step: 0.024365 s/step

Evaluating 'PD_JOINT_DELTA_POS_VEL'
16 envs	  -> t_elapsed: 2.065841 seconds | t_elapsed_per_step: 0.020658 s/step
32 envs	  -> t_elapsed: 2.544216 seconds | t_elapsed_per_step: 0.025442 s/step
64 envs	  -> t_elapsed: 2.933631 seconds | t_elapsed_per_step: 0.029336 s/step
```

Created with:
``` python
import gymnasium as gym
import mani_skill.envs
from time import time

def time_env(env):
    env.reset(seed=0) # reset with a seed for determinism
    num_steps = 100
    action = env.action_space.sample()
    t0 = time()
    for _ in range(num_steps):
        env.step(action)
    t_elapsed = time() - t0
    env.close()
    return t_elapsed, t_elapsed / num_steps

for controller in [
    "pd_joint_delta_pos",
    "pd_joint_pos",
    "pd_ee_delta_pos",
    "pd_ee_delta_pose",
    "pd_ee_pose",
    "pd_joint_target_delta_pos",
    "pd_ee_target_delta_pos",
    "pd_ee_target_delta_pose",
    "pd_joint_vel",
    "pd_joint_pos_vel",
    "pd_joint_delta_pos_vel",
]:
    ENV = f"PokeCube-v1"
    print(f"\nEvaluating '{controller.upper()}'")
    for num_envs in [16, 32, 64]:
        print(f"{num_envs} envs", end="\t")
        env = gym.make(ENV, obs_mode="rgbd", control_mode=controller, num_envs=num_envs)
        t_elapsed, t_elapsed_per_step = time_env(env)
        print(f"  -> t_elapsed: {round(t_elapsed, 6)} seconds | t_elapsed_per_step: {round(t_elapsed_per_step, 6)} s/step")
```

Thanks!
- jeremy

StoneT2000, 2025-03-21:
with GPU sim non delta controllers  for end effector control use pytorch kinematics and tries to solve to pretty low error to get the joint actions that achieve the desired pose as close as possible.

I think the default config for it is to use 200 iterations max to solve the IK problem. I haven't exposed directly an API to modify that but we can (see the kinematics.py file)

Slowdown is probably due to that. Since it's just implemented in torch perhaps it has some slowness in it. I'm sure a cuda kernel would be far better.

Delta end effector control is quite fast because we simply compute a jacobian and then use a pseudo inverse delta control method that is approximate (but pretty inaccurate actually at certain rotations). This is the same setup as most other gpu sims i think (i last checked isaac lab a few months back and this is what they do as well usually)

I am not sure why ee target delta pose control is so slow relatively, should be same speed as non target delta ee pose

jstmn, 2025-03-21:
Got it, thanks for clarifying and for the quick response. I didn't realize a complete IK solution was being found - I assumed it was a single jacobian call plus a levenberg-marquardt or jacobian pseudo-inverse update performed.   

I'll try changing the params for the pytorch kinematics call. Because there's only a single timestep to reach for the solution, there's really no need for more than a few iterations. If its not straightforward, then just running the 'PD_EE_DELTA_POSE' controller with delta = target-pose - current-pose would be the straightforward approach

StoneT2000, 2025-03-21:
pseudo inverse should not work for absolute end effector control however?

jstmn, 2025-03-21:
Ah, well I mean you can use the jacobian pseudo-inverse in the IK. Not sure if I was clear but my previous assumption was that the full IK problem wasn't being solved - rather just a few IK iterations were done to find the delta-joint-configuration

jstmn, 2025-03-21:
BTW, I'd recommend converting the existing jacobian pseudo-inverse updates to use the Levenberg-Marquardt method! I've found that LM results in faster and more stable convergence. See this result, generated from here: https://github.com/jstmn/Jrl/blob/master/scripts/batch_ik_covergence_analysis.ipynb

[image]

Implementation is here for reference: https://github.com/jstmn/Jrl/blob/master/jrl/robot.py#L928

StoneT2000, 2025-03-25:
Oh interesting, thanks for the reference! Maybe we should just use LM method instead (I am not well versed in IK solvers / delta IK solvers etc.).

Might not have too much time this or next week to add this, but this would be a great contribution if you have time to help add it

jstmn, 2025-03-26:
I've yet to come across a situation where the non-LM method does better so i'd agree! 

Sounds good about the contribution, I'll write an implementation and get a PR in.

StoneT2000, 2025-03-27:
To get started all you need to do really is just update the kinematics.py file in the codebase with something better. Hopefully it is straightforward to understand (and if you have a better API design I am all ears, I imagine users may want to modify the IK configuration/setup outside of the defaults)

jstmn, 2025-04-19:
Hey just an FYI i'm going to get to this once my Corl push is over. I've been using the ee_delta_pose controller with the following conversion from absolute pose for the time being, which has been working well

``` python
current_ee_pose = _4x4_to_SE3(obs["agent"]["world__T__ee"][env_idx].cpu().numpy())
R_difference = torch.tensor(    root__T__ee_desired.R * np.linalg.inv(current_ee_pose.R), dtype=torch.float32, device=DEVICE)
action_delta[env_idx, 0:3] = torch.tensor(
    world__T__ee_desired.t - current_ee_pose.t, dtype=torch.float32, device=DEVICE
)
action_delta[env_idx, 3:6] = matrix_to_euler_angles(R_difference, "XYZ")
action_delta[env_idx, 6] = gripper_angle_desired

# Increase the delta to make the tracking more aggressive. 0:6 means leaving the gripper alone
action_delta[:, 0:3] *= 7.0  # xyz
action_delta[:, 3:6] *= 1.0  # rpy
```

StoneT2000, 2025-06-12:
Hi @jstmn some people wanted to know how your little script works.

Do you have the code for running this trick / what is the se3 function you are calling exactly?

And can this be batched easily as well?

StoneT2000, 2025-06-12:
This seems to work fine for me on the latest git commit of maniskill. What commit are you on?

jstmn, 2025-06-13:
Hm that's weird. I'm on most recent version of main. I made an issue here: https://github.com/haosulab/ManiSkill/issues/1127

jstmn, 2025-06-19:
OK, i've got a working implementation! https://github.com/haosulab/ManiSkill/pull/1148

Somehow it seems that the current implementation got 0.25809 -> 0.06584 = ~4x faster since I made this post. Either way, my change is still 2-3x faster, and likely more stable.

closed 2025-08-19 by StoneT2000